bigtreetech/Manta-M8P #1: schematic diagram of Manta-M8P

mink007, 2022-07-31:
I am looking for schematic diagram of Manta-M8P.
Thank you.

TheRevenantStar, 2022-08-14:
Schematics are in the [manual](https://github.com/bigtreetech/Manta-M8P/blob/master/BIGTREETECH%20MANTA%20M8P%20V1.0%20User%20Manual.pdf)

Fisheiyy, 2022-09-05:
> Schematics are in the [manual](https://github.com/bigtreetech/Manta-M8P/blob/master/BIGTREETECH%20MANTA%20M8P%20V1.0%20User%20Manual.pdf)

no, they arent, thats a pinout, not a schematic

PeterHouseJr, 2022-09-10:
Is there a schematic available?

The power connections are not well explained in the preliminary manual and I would like to verify connections on the schematic before applying power.

Specifically: do I need to connect the 'Motor Power' and 'Bed Power' in addition to the 'POWER' connections.  It looks like the eight motor drivers are each jumpered to use the 'POWER' connection.  The bed power also seems to be connected to the 'POWER' connections.  If I want to power the bed separately, how much voltage and power can I use and can it me AC or do I need a separate driver for an AC powered bed heater?

A schematic should answer all of these questions.

Fisheiyy, 2022-09-12:
> Is there a schematic available?
>
> The power connections are not well explained in the preliminary manual and I would like to verify connections on the schematic before applying power.
>
> Specifically: [...]

the Motor Power connection is for 48V is you want to run TMC5160 Pro/HV Drivers, they are jumpered by default to run on the 24V In from the normal POWER connection, the Bed Power connection is for 24V in so its not stressing the main POWER connection with so much current, So you need to connect the Bed Power connection if you want to use a Bed, it is DC only and up to a certain amount of current at 24V I think it is 15A, You would need an external AC Driver for an AC Bed.

PeterHouseJr, 2022-09-12:
@Fisheiyy Thank you so much for your very informative response.  On my Manta8P, the 'Bed Power In' and the 'Main Power In' are connected.  There are three small fuses on the board, each rated for 15A.  Their positioning suggest they are for the three power inputs.  I do not see a jumper anywhere to disconnect the 'Bed Power In' from the 'Main Power In'.  I am about to power up the Manta to drive some small test steppers and will post if I find anything different.

For the record: I would still like to see a schematic.

Fisheiyy, 2022-09-12:
> @Fisheiyy Thank you so much for your very informative response.  On my Manta8P, the 'Bed Power In' and the 'Main Power In' are connected.  There are three small fuses on the board, each rated for [...]

So would we all, The bed power is already seperate from the main power, you use a jumper to change the drivers from main power 24V to motor power which is meant for 48V

PeterHouseJr, 2022-09-13:
@Fisheiyy 
On my board, M8P, the Bed Power and the POWER connections are connected.  I don't see an easy way to disconnect them if I wanted to, which I do not at this time.  I am thinking about an AC high power bed for quicker heating but am worried about the high voltage and the quick heating causing temperature gradients.

thisiskeithb, 2022-10-30:
Schematic has been uploaded: https://github.com/bigtreetech/Manta-M8P/blob/master/Hardware/BIGTREETECH%20MANTA%20M8P%20V1.0-SCH.pdf

emersonsc, 2022-11-23:
@mink007 could you marked this as closed to clean up the issues area. Thank you

closed 2022-11-23 by mink007

Alex-LR, 2023-12-10:
Schematic was removed, so no schematics again!