bigtreetech/Manta-M8P #2: Dual Filament Runout Sensors

S1L3NTANG3L, 2022-08-01:
Is there a way to connect two SFS filament run out sensors to the board?

clevelandcs, 2022-10-21:
@S1L3NTANG3L Depends on how much other hardware you need to run. There is only the one dedicated port for their SFS sensor but if you don't need 6 motor endstops (which I'm impressed if you do) you could probably re-terminate the cable for the second SFS to use one of the endstops and just a signal pin from another end stop or other input pin you're not using elsewhere on the board. The pinout diagram in the Hardware folder should be able to put you down the right path.

emersonsc, 2022-11-23:
Really any available I/O pin can be used as @clevelandcs mention. You'll just have to remove the wires from the plug, put the 5v, GRD, and first signal pin into a 3 pin plug for the one available endstop, and the single 2nd signal pin go to the other open endstop. This how ever will ONLY work if you arent using sensorless homing on the endstops/motor your going to use for the 2nd filament sensor.