bigtreetech/Manta-M8P #4: CB1 Wifi

rmilyard, 2022-08-09:
So I wrote the CB1 image.  

In the system.cfg file I put in my Wifi info. I did the 2.4ghz since not sure if 5 supported.  I am looking for IP and not finding it on network.  On the Manta board there is a soild RED LED near the CPU and green LED blinking near the CB1.

CTechLite, 2022-09-09:
Same here, did BTT ever get back to you?

lowtolerance, 2022-09-10:
I am also having this issue. After about 30 seconds from applying power, the “ACT” light starts blinking.

clevelandcs, 2022-10-21:
@rmilyard Have you resolved your issue? I can confirm that it will only connect to 2.4GHz networks (it even doesn't like networks support both on the same SSID) but using a dedicated 2.4GHz network and the 2.2.0 (Kernel 5.16) release from September 29th has been working fine for me with WiFi signal within margin of error when compared to my rpi 3B+ boards.

pyro159, 2023-04-21:
I ran into a similar issue and found something that worked for me. I changed the name of my wifi to a shorter/simpler name ("Gaming" in my case) and got it to connect correctly. It may have helped, but I used balanEtcher instead of Rasberry PI Imager to flash the drive. I will edit my post if I run into any anomalies

Doormat1, 2023-04-28:
The instructions in the manual dont work however this does - https://www.makeuseof.com/connect-to-wifi-with-nmcli/

sxming1010, 2024-01-31:
WIFI设置要保持和自己路由器上网WFI名称及密码一样就好，注意和marlin的设置有所区别，最容易搞错的点。

wadware, 2024-07-16:
Hi, after connecting the wifi antenna to the CB1, it logged into the network and is already working.