bigtreetech/Manta-M8P #5: Use with external Pi (or other system)

TheRevenantStar, 2022-08-14:
Since there are a number of users unable to use the CB1 and the CM4 is under shortage, is there a way to use the M8P with a regular Pi over USB?

I'd like to connect the Pi 4B I used in an older setup to connect to this board but I can't figure out how r if this is possible.
Please advise.

PeterHouseJr, 2022-09-19:
I would also like to use an external Pi to run the Manta 8P.  How can I get access to the STM32 serial connection?

NAPCAL, 2022-10-02:
The STMG0B1 USB interface is directly connected to the CB1/CM4 high density connector.

Iceman231173, 2022-10-23:
What does that mean?  can the board be used with external Raspberry Pi or not?

PeterHouseJr, 2022-10-23:
There is no serial port or USB port to communicate with the STM32 other than the embedded CM4 or CB1 modules.

The serial port from the STM32 microntroller is implemented through a usb port using two port pins from the STM32 directly connected to the high density connectors of the CM4/CB1 Pi module. leaving no way to connect an external Pi or other computer to the Manta8P.  

I have thought about making a PCB which would plug into the CM4/CB1 location and provide a USB connector for an external connection.  Unfortunately, I cannot tell if there is any interest to sell these although I may just do it for myself.

closed 2022-10-31 by TheRevenantStar