bigtreetech/Manta-M8P #6: output_pin PS_on config ?

Kostt3d, 2022-08-15:
Bonjour
Je rencontre un probleme avec la configuration du relay de fin d'impression. BTT relay v1.2. il s'arrete quand j'allume l'imprimante avant que le mcu ne soit chargé.
Dans le fichier printer.cfg j'ai mis les informations suivantes : 

[output_pin ps_on_pin]
pin: PC3
value: 1
shutdown_value: 0

quel est la configuration pour que cela fonctionne ?

emersonsc, 2022-11-23:
https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/blob/master/BIGTREETECH%20Relay%20V1.2/BIGTREETECH%20Relay%20V1.2/Hardware/%E5%BC%80%E5%85%B3%E5%8A%A0%E6%89%93%E5%AE%8C%E5%85%B3%E6%9C%BA.jpg

I know thats for a different MCU, but it should give you an idea on how to wire it correctly to function properly. Only other thing you could try is to try ^PC3 and see if that helps.

GregNuspel, 2022-12-10:
I have the same problem with my printer. The pin is configured as:
```
[output_pin PS_ON_OFF]
pin: PC3
value: 1
```
My print end code is:
```
[gcode_macro END_PRINT]
gcode:
    # Turn off bed, extruder, and fan
    M140 S0
    M104 S0
    M106 S0
    # Move nozzle away from print while retracting
    G91
    G1 X-2 Y-2 E-3 F300
    # Raise nozzle by 10mm
    G1 Z10 F3000
    G90
    G1 X10 Y330 F1000
    # Disable steppers
    M84
    G90 ; absolute positioning
    TEMPERATURE_WAIT sensor=extruder maximum=60
    SET_PIN PIN=PS_ON_OFF VALUE=0; shut down the printer
```

The board boots too slow for the 15 second time delay of the relay board. I have tested the relays actions for the end of print and it works as predicted. I need to know if there is a way to make PC3 go high before Klipper loads or is there a way to lengthen the 15 second delay on the relay board?

jariddler, 2023-04-17:
When building the firmware if you set pin pc3 high on startup it will fix the issues with the relay board

GregNuspel, 2023-05-06:
I tried to enter this in menuconfig : `set_digital_out pin=PC3 value=1`

When I ran make I had the following error: Unknown initial pin `set_digital_out pin=PC3 value=1`

Is there a different pin callout on the microprocessor level?

jariddler, 2023-05-06:
Just put the pin. Pc3 nothing else

Kostt3d, 2023-05-10:
Hy everybody
i switched my motherboard for a M4P and this setup work perfectly
[output_pin ps_on_pin]
pin: PD1
#   The pin to configure as an output. This parameter must be
#   provided.
#pwm: False
#   Set if the output pin should be capable of pulse-width-modulation.
#   If this is true, the value fields should be between 0 and 1; if it
#   is false the value fields should be either 0 or 1. The default is
#   False.
#static_value:
#   If this is set, then the pin is assigned to this value at startup
#   and the pin can not be changed during runtime. A static pin uses
#   slightly less ram in the micro-controller. The default is to use
#   runtime configuration of pins.
value:1
#   The value to initially set the pin to during MCU configuration.
#   The default is 0 (for low voltage).
shutdown_value : 1

closed 2023-07-21 by Kostt3d

ajaytanna, 2024-05-12:
> Just put the pin. Pc3 nothing else
As in attached picture?
[image]