bigtreetech/Manta-M8P #9: SPI1 communication

stcs1981, 2022-08-25:
Hello, I have "Global scaler" communication problems with various brands of TMC5160 drivers. In the examples, there are some pins in the table, there are others. I have tried several combinations and there is no way!
This is the configuration

## X Stepper on Motor1(B Motor)
[stepper_x]
step_pin: PE2
dir_pin: !PB4
enable_pin: !PC11
endstop_pin: sht42:^PA0 
rotation_distance: 32
microsteps:16                
full_steps_per_rotation: 200
position_endstop:-11.5
position_max: 310             
position_min: -11.5              
homing_speed: 100             
homing_retract_dist: 5       
homing_positive_dir: false

[tmc5160 stepper_x]              
cs_pin: PC10
spi_bus: spi1
#spi_software_sclk_pin: PA5
#spi_software_mosi_pin: PA7
#spi_software_miso_pin: PA6
interpolate: True
run_current: 2
#stealthchop_threshold: 0
#sense_resistor: 0.075

Thanks in advance!!!

GregNuspel, 2022-11-11:
I'm having similar problems with the TMC2130 SPI config.

stcs1981, 2022-11-11:
Hello dear friend this is my configuration!!!

GregNuspel, 2022-11-11:
The link isn't working for me. Can you post your config here. 
Thanks

stcs1981, 2022-11-11:
[stepper_x]

step_pin: PE2

dir_pin: PB4

enable_pin: !PC11

microsteps: 16

rotation_distance: 32

#endstop_pin: ^PF3

full_steps_per_rotation: 200

position_endstop: -5

position_min: -5

position_max: 310

homing_speed: 150

homing_retract_dist: 5

[tmc5160 stepper_x]

cs_pin: PC10

spi_software_sclk_pin: PA5

spi_software_mosi_pin: PA7

spi_software_miso_pin: PA6

run_current: 2

GregNuspel, 2022-11-11:
Thanks I hadn't put in the software spi info.

stcs1981, 2022-11-11:
I went crazy and in the end this was the only way to make spi bus work!! Greetings and I hope I have helped you fellow sufferer!!!!

emersonsc, 2022-11-22:
Can confirm this issue as well. Simply providing the "cs" pin and the spi bus will not work. You MUST specify each pin, not just the CS pin in order to use drivers in spi mode.

spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6

InBlainSight, 2023-04-03:
I also had the problem and switching over to motor power did the fix. For some reason regular board power didn't work. I used another psu but I don't think that matters. Global scaler was the issue as well. I even used the same voltage as the board power

stcs1981, 2023-04-03:
It's solved!!! For a while now if you want I can send you the configuration when I get home later

stcs1981, 2023-04-04:
> I would be very happy to get the solution to this problem.
>
> Thanks
>
> -Greg

[stepper_x]
step_pin: PE2
dir_pin: PB4
enable_pin: !PC11
microsteps: 16
rotation_distance: 32
#ndstop_pin: ^PF3
full_steps_per_rotation: 200
position_endstop: 0
position_min: 0
position_max: 346
homing_speed: 200
homing_retract_dist: 5

[tmc5160 stepper_x]
cs_pin: PC10
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
run_current: 2

tstone, 2023-05-13:
For anyone who finds this issue in the future, I verified this does work with the M8P V1.1 as well.

NOTE: Because of the way the SPI bus works, all drivers using SPI must be configured as mentioned above with the 3 pins. The problem will persist if even just 1 of the drivers is still using `spi_bus`.

kymlalu, 2024-06-19:
> For anyone who finds this issue in the future, I verified this does work with the M8P V1.1 as well.
>
> NOTE: Because of the way the SPI bus works, all drivers using SPI must be configured as mentioned [...]

For M8P V1.1 HW SPI works but you have to specified spi bus speed, default one is too fast. I started somwhere on low side and tried to go up until i got error.
M8P V2 hoewer does't have correct pin's for HW SPI so SW SPI is the only options. ( Pins PG6,7,8 of STM32H7xx doesn't have SPI function, why someone designed it in this way is mystery to me )

PeterXQC, 2024-09-21:
> For M8P V1.1 HW SPI works but you have to specified spi bus speed, default one is too fast. I started somwhere on low side and tried to go up until i got error. M8P V2 hoewer does't have correct [...]

Can you elaborate on how you configured it for M8P V2? I am doing 

[tmc5160 stepper_x]
cs_pin: PE3
spi_software_sclk_pin: PC10
spi_software_mosi_pin: PC12
spi_software_miso_pin: PC11
run_current: 2

This is giving me error. I think the pins are correct but I'm not super sure. I also have EBB 36 can with another use of spi bus:

heater_pin: EBBCan: PB13
sensor_type: MAX31865
sensor_pin: EBBCan: PA4
spi_software_sclk_pin: EBBCan:PA5
spi_software_mosi_pin: EBBCan:PA7
spi_software_miso_pin: EBBCan:PA6
rtd_nominal_r: 1000
rtd_reference_r: 4300
rtd_num_of_wires: 2
min_temp: 0
max_temp: 350

Would this mess with the stepper drivers' spi config?