bigtreetech/Manta-M8P #10: Dual Motor2 or Motor3? Unclear

IonCaza, 2022-08-27:
Manual (page 6) point 17. references `Motor Drive Interface: Motor1, Motor2(dual motor interface), Motor3, ...` however, schematic and board have "M1", "M2", "M3A" and "M3B". So which one is dual motor? Motor 2 or Motor 3?

basha318i, 2022-09-03:
Motor 3; since it says M3A & M3B.

IonCaza, 2022-09-03:
> Motor 3; since it says M3A & M3B.

Lol you seem to be very confident of the engravings / schematics pictures... all the other literature (including manual) has references to 'Motor 2 (dual motor)'

clevelandcs, 2022-10-21:
@IonCaza I don't know if they changed it at some point but when I got my boards a few weeks ago the manual matches the schematics in the hardware directory. Also in my adventure in wiring and rewiring a number of times I can confirm from experience that it is Motor 3 that is the dual motor outputs.

Okamimann, 2023-05-28:
It's a typo in the Introduction section. I have a Manta M8P v1.1, and that specific point you mention has been updated to say Motor1, Motor2, Motor 3(dual motor interface) ... but the truth is, that technically doesn't matter. You can SEE where the dual plug is on the board (next to the 3rd slot for stepper drivers), and you can look at both the Interface Diagram (page 7 on the 1.0 manual, page 10 on the 1.1 manual) and the the Pin-Out diagram (page 8 on the 1.0 manual, page 10 on the 1.1 manual for the 1.0 board and page 11 on the 1.1 manual for the 1.1 board). Just make sure you get the correct pins in the correct settings locations, and you're fine.

For what it's worth, another way to look at it would have been to realize that just about every manufacturer that I know of, if they put the plugs for the motors in a row, the expected behavior is X, Y, Z, Z1, E. The "default" (but commented out) settings for motors 6, 7, and 8 are all for extruders as well (4 extruders?! What, are you creating your own Palette machine? On second thought ... that might be useful actually). But all of the motors can be switched around or used for other purposes, you just have to make sure you put the correct pins in the correct section. Accidentally plugged the X motor into the Y slot and vice versa, and now the board is in a difficult location so you don't want to tear everything apart just to fix the wires? Simply swap the pins in the printer.cfg settings file and you're good to go.