bigtreetech/Manta-M8P #13: Using an external USB-connected Pi

lowtolerance, 2022-09-25:
I know the Manta is built to accept a Raspberry Pi CM 4 module, but is it possible to use a Pi connected to the Manta by USB instead of using the CM4 or CB1 modules? I can’t get my hands on a CM4 module and the CB1 module I received has major issues that I’m trying to get that sorted out through BTT’s customer support. 

I tried the obvious of hooking up to the Manta’s USB-C port, but it’s not being recognized as a serial device. Is the USB-C just for flashing the MCU in DFU mode?

NAPCAL, 2022-10-02:
The USB-C is only connected to the CB1/CM4 high density connector.

The STM32G0B1 USB connection is directly connected to the USB hub chip for the CB1/CM4.

Fisheiyy, 2022-10-17:
> The USB-C is only connected to the CB1/CM4 high density connector.
> 
> The STM32G0B1 USB connection is directly connected to the USB hub chip for the CB1/CM4.

and where are you getting this information? there is no schematic so therefore no way to really know

NAPCAL, 2022-10-17:
If you load Klipper onto the M8P/M4P, you will never see the M8P/M4P on the USB-C port; you will from the CB1/CM4.

Typically an RPi has four USB ports; the M8P/M4P has two plus a USB header behind the two USB ports. This leaves one that you always see the M8P/M4P on when doing a lsusb on the CB1/CM4, also displayed are the physical connection numbers 1-4, the M8P/M4P will always be on the same physical connection.

Iceman231173, 2022-10-23:
What does that mean?  can the board be used with external Raspberry Pi or not?

NAPCAL, 2022-10-23:
Without using a CB1/CM4 first to set up the M8P firmware for UART, the M8P USB option can not be accessed since it is hardwired to the CB1/CM4 high density connector with no external connections.

Fisheiyy, 2022-10-23:
> What does that mean? can the board be used with external Raspberry Pi or not?

you have to use the display pins or some extra motor driver pins and set the firmware to use UART then you can connect the pi to manta, your probably better off buying a cm4 or cb1

NAPCAL, 2022-10-23:
> > What does that mean? can the board be used with external Raspberry Pi or not?
> 
> you have to use the display pins or some extra motor driver pins and set the firmware to use UART then you can connect the pi to manta, your probably better off buying a cm4 or cb1

That is basically the idea, the board wasn't designed to be used without a CB1/CM4. To be able to currently fully use the GPIO, Camera, DSI you would need to use a CM4.

non-bin, 2025-05-27:
Is this still the case for the v1.1?