bigtreetech/Manta-M8P #17: cannot access '/dev/serial/by-id': No such file or directory

hippieua, 2022-10-30:
release [V2.2.0](https://github.com/bigtreetech/CB1/releases/tag/V2.2.0)

[Hurakan_Debian11_Mainsail_kernel5.16_20220929.img.xz](https://github.com/bigtreetech/CB1/releases/download/V2.2.0/Hurakan_Debian11_Mainsail_kernel5.16_20220929.img.xz)

manta m8p board with CB1 module

biqu@BTT-CB1:~/klipper$ ls /dev/serial/by-id
ls: cannot access '/dev/serial/by-id': No such file or directory

biqu@BTT-CB1:~/klipper$ uname -a
Linux BTT-CB1 5.16.17-sun50iw9 #2.2.0 SMP Wed Sep 28 15:50:52 HKT 2022 aarch64 GNU/Linux

hippieua, 2022-10-30:
bad connection CB1 to Manta M8P board. Just push it harder )

closed 2022-10-30 by hippieua

wodier12, 2023-10-21:
what does "just push it harder mean?"

hippieua, 2023-10-21:
> what does "just push it harder mean?"

laterally, just i just pushed cb1 harder to m8p until here it clicks in socket

wodier12, 2023-10-21:
hmmmmm

wodier12, 2023-11-02:
" just push it harder " does not work, but thanks for trying...