bigtreetech/Manta-M8P #18: Unable to flash firmware

airbornetrooper82573, 2022-11-12:
I've tried via SD card and it never changes firmware.bin to cue. I have flashed this board once but I am getting some errors and attempting to flash again. Getting the following errors:

`usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00
biqu@BTT-CB1:~/klipper$ make flash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00
  Flashing out/klipper.bin to /dev/serial/by-id/usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00
Entering bootloader on /dev/serial/by-id/usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00
Device reconnect on /sys/devices/platform/soc/5200000.usb/usb2/2-1/2-1.4/2-1.4:1.0
sudo dfu-util -p 2-1.4 -R -a 0 -s 0x8002000:leave -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08002000, size = 26644
Download        [=========================] 100%        26644 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status

Failed to flash to /dev/serial/by-id/usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00

make: *** [src/stm32/Makefile:90: flash] Error 255`

Board is connected to 24v power supply on my Voron 2.4

airbornetrooper82573, 2022-11-12:
Disregard the Make Flash error... I see at the very bottom of page 37 it says to disregard the error. I am still unable to flash with SD card but I guess it is fine.

NiLMoXiE, 2022-11-13:
I am having trouble flashing my M8P via SD card - it just doesn't work - I cant do it by USB either as I can't get my board ID, any tips or hints?

airbornetrooper82573, 2022-11-13:
When I did mine originally I had the jumper set and powered via usb. If you can get into Mainsail/Fluidd you should pull the ID like the guide says.

NiLMoXiE, 2022-11-13:
Thanks man, I have just realised there are 2 SD card slots, I was using the wrong one.

closed 2022-11-19 by airbornetrooper82573