bigtreetech/Manta-M8P #19: DSI and CSI Connection

Alvin5311, 2022-11-15:
I have the CM4 (8Gb RAM, 8Gb eMMC) on my M8P board and I've connected the BTT Pi TFT50 v2.0 and the Picam 8MP v2.0 to the DSI and CSI FPC on the bottom of the board and got no display and picture. Need help to troubleshoot.

Thanks.

Alvin5311, 2022-11-19:
Nevermind, the type B FFC ribbon cable doesn't work for the M8P board with the Pi TFT50 v2.0 display, I had to swap it with a type A ribbon cable.

closed 2022-11-19 by Alvin5311

aywosh, 2022-11-21:
Hi Alvin5311, did you find a solution because you closed the issue? I have the same issue. Thanks.

TheLexus, 2023-11-15:
@Alvin5311 Im to try to use the M8P with a DSI 5" display which worked with a pi without problems before. But im a bit lost on how to use the dsi connector on the underside of the board... I cannot get the cable in it. What do I maybe wrong? Can you give me a hint from which side the cable plugs in... I tried it from both sides but it fits only from the "outside" where the screw hole is... but the display is not recognised by linux seen from the dmesg. It even does not light up, so I guess I made something wrong.

TheLexus, 2023-11-19:
Ok, was able to fix it up. In the end the cable connectors were problematic. be sure that you really insert them as much as possible. On the Manta the pins need to face up (so that you see the silver plates). Verify that you use a Type-B cable which has the contacts on both ends at the same side. I used the fkms overlay instead of the kms one and no additional one for the screen. But my screen isn't from waveshare, so maybe yours different.