bigtreetech/Manta-M8P #21: Probe Config Issue (B1 SE Plus Conversion)

mario54bro, 2022-11-23:
I'm working on trying to configure the probe on the B1 SE Plus on the Manta M8P board. The blue (and Green) light on the FX712 board flashes randomly. This is trigging the Z Endstop before it reaches the bed. 

I used the Example setup from the B1 SE Plus and here is what I have for the config.

[probe]
pin: ^!PB1
z_offset: 0.0
x_offset: 0.0
y_offset: 0.0
speed: 10.0
samples: 2
samples_result: average
sample_retract_dist: 2.0
samples_tolerance: 0.2

[output_pin probe_enable]
pin: PB2
value: 1

emersonsc, 2022-11-23:
have you tried putting the pullup on the output pin, ex pin: ^PB2?

You should go to the machine tab in Mainsail, then scroll down and you can query the probes. Keep in mind its not real-time, and has to be refreshed every time you want to check them. Its a useful tool when getting the settings right for your endstops and probes.

mario54bro, 2022-11-23:
I have tried this, it causes this Error
Invalid pin description '^PB2'
Format is: [!] [chip_name:] pin_name

However, when this error is triggered, the light doesn't blink..

emersonsc, 2022-11-23:
> I have tried this, it causes this Error Invalid pin description '^PB2' Format is: [!] [chip_name:] pin_name
> 
> However, when this error is triggered, the light doesn't blink..

How about just ! OR changing the 1 to 0?

mario54bro, 2022-11-23:
Tried all those combinations with no luck.

mario54bro, 2022-11-24:
I have found that plugging the probe 5v and ground into a different supply source (not on the board) resolves the issue. Any idea why this would be?

emersonsc, 2022-11-24:
> I have found that plugging the probe 5v and ground into a different supply source (not on the board) resolves the issue. Any idea why this would be?

Have you tried another 5v pin on the board?

mario54bro, 2022-11-24:
> > I have found that plugging the probe 5v and ground into a different supply source (not on the board) resolves the issue. Any idea why this would be?
> 
> Have you tried another 5v pin on the board?

Yes, tried multiple other 5v pins on board with no luck.