bigtreetech/Manta-M8P #22: BTT Mini 12864 doesn't work

Bill-Dung, 2022-11-24:
The Neopixel of the 12864 Mini are working fine but the screen itself stay blank. I used following Pins:

`[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PE9, EXP1_2=PE10,
    EXP1_3=PE11, EXP1_4=PE12,
    EXP1_5=PE13, EXP1_6=PE14,    # Slot in the socket on this side
    EXP1_7=PE15, EXP1_8=PB10,
    EXP1_9=<GND>, EXP1_10=<5V>,

    # EXP2 header
    EXP2_1=PB14, EXP2_2=PB13,
    EXP2_3=PF7, EXP2_4=PB12,
    EXP2_5=PE7, EXP2_6=PB11,      # Slot in the socket on this side
    EXP2_7=PE8, EXP2_8=<RST>,
    EXP2_9=<GND>, EXP2_10=<NC>`

and those klipper config entries:

`[display]
lcd_type: uc1701
cs_pin: EXP1_3
a0_pin: EXP1_4
rst_pin: EXP1_5
contrast: 63
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1

[neopixel fysetc_mini12864]
pin: EXP1_6
chain_count: 3
color_order: RGB
initial_RED: 0.05
initial_GREEN: 0.5
initial_BLUE: 0.5
`

The display is working fine at a Manta M4p but not at the M8P. I tried different cable in different directions of the plugs, no way.

turpinator, 2022-11-26:
It looks like you havent defined the SPI pins
`spi_software_miso_pin: EXP2_1
spi_software_mosi_pin: EXP2_6
spi_software_sclk_pin: EXP2_2`

Bill-Dung, 2022-11-30:
That's it, thanks. I would never have thought of it, since the same display runs on the M4P without these lines. Doesn't matter, the main thing is that it works now. Thx again.

closed 2022-11-30 by Bill-Dung