bigtreetech/Manta-M8P #25: No serial connection between MCU and CB1

Lzingano, 2022-12-13:
Hello all, I am also writing it here, together with Reddit.

I just received my Manta M8P and CB1, but am experiencing a bit of an issue here. 

- I installed the release V2.2.1 image from [here](https://github.com/bigtreetech/CB1/releases) to the CB1 and all went pretty well.
- I used BTT's instructions to compile the firmware for the MCU and it also seems to have worked (and I double checked the klipper configuration to build the .bin). 
- I have tried powering the board using the USB-C with the VUSB jumper on, as well as with 24V PSU and no VUSB jumper. The difference I saw is that LEDs get lit a little bit brighter with 24v, but that's it. 
- SD cards are also in the correct position. 
- I have updated all the CB1 software before building the .bin. The MCU seems to be flashing correctly.
- Also, I can connect to the CB1 without any issues via LAN or Wifi, which I use to SSH into it. The Webapp works as well. 
- ACT light flashed, CB1 LED is lit and

What I can not have working is the serial connection between both MCU and CB1. It looks like an issue with the USB somewhere in the chain. The USB chip is not overheating or anything, but when I connect a flash drive to any USB port, it isn't recognized. 
Does anyone have a clue what could be happening? I fear that I might have a deffective usb hub chip somehow, which would be pretty bad.
[image]

Any help is appreciated =)

TIA

jeronbernal, 2022-12-27:
waiting for a solution to this aswell

Lzingano, 2023-01-01:
I am in contact with BTT, but they have so far not given any solution. I will this week make a warranty claim, because the board is probably toast.

jfneptune, 2023-01-06:
Did you ever find out what the issue was?  Was it the board being toast or something else?

jeronbernal, 2023-01-06:
I ended up reinstalling mainsail and klipper doing a fresh install and it
started working again

jfneptune, 2023-01-06:
How did you reinstall everything?  Did you reformat the microsd card and follow the instructions in the BTT manual or install some other way?

Lzingano, 2023-01-06:
I am also interested in this. I haven't found a solution yet. At this moment, I requested to return the card and will probably get a new one after this and hope it works. They are probably shipping the rev 1.1 right now (SKU changed).

evplasmaman, 2023-01-17:
I'm also seeing this issue. I just got my V1.1 MP8 board with a CM1 but no luck with the serial connections on 2.2.1. I've even tried going down to the 2.1.2 image to see if that makes any difference but no luck.

Can anyone else confirm that their MCU LED is not active after flashing the firmware generated by klipper? I only see it blink when I try to load a new MCU firmware or if I reset the chip. Not sure if that is a cause or a symptom of the serial issue.

I might try to see if I can follow in @jeronbernal 's footsteps but I'm new to klipper and not quite sure where to begin with doing a fresh install.

emersonsc, 2023-01-23:
When you compile the firmware for the board, are you choosing USB or serial connection? The M8P & CM1 uses the USB connection option.

evplasmaman, 2023-01-23:
Yes for communication interface I selected (USB (on PA11/PA12))

I've done some more digging in to the issue and it looks like the FE1.1s chip is only outputting 0.9V from its internal voltage regulator when it's supposed to be 3.3V. Applying 3.3V from another current limited source (which the chips data sheet says you are allowed to do) makes all the USB devices show up and the board work normally. 

I'm not sure if this is due to the chip having a bad internal voltage regulator or of this stems from an issue on the new V1.1 boards. I contacted BTT yesterday and will see what they say.

If someone feels brave enough to test their board for the same issue and they have a fine enough probes for their multi meter here is where I checked the voltage supply with the ground connected to one of the supply GND pin on the other side of the board.
[image]

Johnnyyr, 2023-02-02:
[image]

I think i have the same issue with manta m5p and cb 1 2.2.1

evplasmaman, 2023-02-03:
> When you compile the firmware for the board, are you choosing USB or serial connection? The M8P & CM1 uses the USB connection option.

I am compiling with the USB option.

evplasmaman, 2023-02-03:
> [image]
> 
> I think i have the same issue with manta m5p and cb 1 2.2.1

Could you try plugging in another USB device and seeing if it shows up in lsusb? With the issue I had I was only able to see the Linux Foundation devices.

Johnnyyr, 2023-02-03:
> > [image]
> > I think i have the same issue with manta m5p and cb 1 2.2.1
> 
> Could you try plugging in another USB device and seeing if it shows up in lsusb? With the issue I had I was only able to see the Linux Foundation devices.

No i can´t but the CB1 is onboard of the m5p. I have flashed the m5p normally without can jeztztz I see it. Unfortunately I do not know how to flash it correctly for Can.

adjustinthings, 2023-02-04:
https://github.com/bigtreetech/CB1#v221-os-setting

bddanford, 2023-02-21:
Swapped out my RP4B for a CB1/PI4 adapter and I cant see any of my USB services. LSUSB shows just the hubs and nothing else, not the camera or the SKR 1.4. I wanted to play around with the CB1 and put the RP4B in my Voron 2.4, which has a RP3b+ in it now. 

Anyone figure this out? I am powering the CB1/RP4 board with a 5v meanwell PSU connectioned to the 5v and ground on the GPIO pins.

jeronbernal, 2023-02-22:
did you enable usb hub via the commands in the manual?

bddanford, 2023-02-22:
You only do that if you want the USB C port to actually be a USB port and not just a power port. I ended up reseating the CB1 and everything showed up just fine. Must not have been seated correctly.

sbuchana, 2023-03-02:
I was pulling my hair out over what might be the same issue. My issues disappeared as soon as I disconnected my mini 12864 display. It turns out that Fysetc wires their display connectors reverse of Biqu. I found this reference that talked about rotating the PS1/PS2 connectors. Solved my issue at least.

https://docs.vorondesign.com/build/electrical/mini12864_klipper_guide.html

Lzingano, 2023-03-02:
So, to give a feedback on my side. I had the board replaced by BIQU. They sent me a Manta M8P V1.1 and it works like a charm with the same CB1. I had the issue that was pointed out [here](https://github.com/bigtreetech/Manta-M8P/issues/25#issuecomment-1400974408). I am not sure what the problem was, but seems like it was a bad board. Maybe if someone is having this same issue, I suggest getting in touch with BIQU. They are willing to help =)

Lazybones2319, 2023-04-27:
I'm having a similar issue with my m8p v1.1 it seems to boot everything but there is no power led near the hdmi ports the photo is after 20 ish minutes trying to boot.  [image]

punjabik, 2023-05-22:
> I was pulling my hair out over what might be the same issue. My issues disappeared as soon as I disconnected my mini 12864 display. It turns out that Fysetc wires their display connectors reverse of [...]

So much thank you for this. You saved me a lot of time. I was unable to get firmware to the manta and it was because fysetc connected display.

keamer, 2023-05-30:
@Lazybones2319 --
Not sure, but did you ever switch back the DIP switches? They switches 3 and 4 seem to be on.

From the manual, page 28:
> 5. When the writing is completed, push the DIP switch 4 (USB OTG) and 3 (RPIBOOT) back to OFF after power off, and power on again to enter the normal working mode.

Arcanengineer, 2023-07-06:
[image]

Lazybones2319, 2023-07-06:
@keamer 

It turns out my issue was a bad sd card. 
I posted this then thought hey I have a spare card I'll flash it with that. 
Low and behold the thing started working and has around 2000 hours run time so far

Arcanengineer, 2023-07-06:
> @keamer
> 
> It turns out my issue was a bad sd card. I posted this then thought hey I have a spare card I'll flash it with that. Low and behold the thing started working and has around 2000 hours run time so far

Not for me

Arcanengineer, 2023-07-06:
Guys... I got my problem solved. For some weird reason, when i pulled the cables of my adxl345 off the board, the serial connection instantly became available. 
So, everyone having this kind of issue should try and see if there is any connections that may be scrambling the firmware...
   I measured vibration, made like 3 or 4 prints before it started to get rusty and not want to restart firmware, until it didnt and i could not have the serial connect by any means.
   I tried to reflash the board firmware, reinstaled debian on cb1, try many different sd cards, but what fixed in the end was pull out some plugs
Hope it helps others

robcampman, 2023-09-11:
I also have a problem with my board. Unfortunately the resistors measure 1.59 volts.  Thus same problem as the po.  Does anyone know how to fix this problem. I would be comfortable soldering or rerouting some wires.

ibrahimahmad-id, 2024-01-15:
> I'm having a similar issue with my m8p v1.1 it seems to boot everything but there is no power led near the hdmi ports the photo is after 20 ish minutes trying to boot.

You mean the ACT led?

I find that if you flash newer CB1 system image, like v2.3.3 either Klipper or Minimal version, the ACT led doesn't lit up/blinking, but if I flash older CB1 system, like the v2.1.2, the ACT led lit up & blinking as usual. Despite that, I still can SSH to my CB1 and open the Klipper UI even if the ACT led doesn't lit up like usual. You can try flash to older CB1 system and see that the LED will lit up again.

Note that I do have similar issues with the rest of the guys here where after months of printing, my CB1 suddenly lost its MCU path to my M8P and can't find it by serial/by-id/.

ploucandco, 2024-01-16:
> Yes for communication interface I selected (USB (on PA11/PA12))
> 
> I've done some more digging in to the issue and it looks like the FE1.1s chip is only outputting 0.9V from its internal voltage regulator when it's supposed to be 3.3V. Applying 3.3V from another current limited source (which the chips data sheet says you are allowed to do) makes all the USB devices show up and the board work normally.
> 
> I'm not sure if this is due to the chip having a bad internal voltage regulator or of this stems from an issue on the new V1.1 boards. I contacted BTT yesterday and will see what they say.
> 
> If someone feels brave enough to test their board for the same issue and they have a fine enough probes for their multi meter here is where I checked the voltage supply with the ground connected to one of the supply GND pin on the other side of the board. (Just be very cautious not to short to the caps to the left of those resistors, that's GND) [image]

Hello, were you able to solve the issue this way? I got a Manta M8P V1.0 and got the same issue with the USB hub not working. I was thinking of connecting pin 20 (VD33F) to the 3.3V board supply as this seems to be the way to get this chip supplied by 3.3V. I just wondered about disconnecting the 5V... Any experience if that works?

PixelVengeur, 2024-02-01:
> Yes for communication interface I selected (USB (on PA11/PA12))
>
> I've done some more digging in to the issue and it looks like the FE1.1s chip is only outputting 0.9V from its internal voltage [...]

I'm having the same symptoms exactly. No USB device is detected, be it the M8P, a flash drive, keyboard, mouse, whatever. I'm not getting 0.9V, mor ein the realm of 1.7V, but it shows teh same issues. I do not have any external PSU I can generate 3.3V from, so I can't confirm whether that would fix the problem.

Claudiuene, 2024-09-18:
hello, I had the same problem with version 2.0 M8P. look here the solution.
[image]
[image]