bigtreetech/Manta-M8P #27: Manta M8P Boot Loop on SPI1 with ADXL345

NiLMoXiE, 2022-12-21:
Connecting an ADXL345 to SPI1 on my Manta M8P causes the MCU to not boot. It seems like the bootloader is conflicting with some other device on SPI1 on boot.

Can anyone confirm if there is any other device on this board that uses SPI1 which would cause this?

Also is there any way to solve this issue - is there a different bootloader I can use? 

Thanks

NiLMoXiE, 2022-12-22:
Despite the update to the user manual for the ADXL345 - it still produces the same issue - MCU will not boot with ADXL345 connected via SPI1

Bischi1997, 2023-01-20:
I tried it, for me it seems to work with Klipper and a CB1, when specifying ports without using SPI1:
```
[adxl345 hotend]
cs_pin: PB15 #PC4 did not work for me, even though I have v1.1 without the 31865
spi_software_sclk_pin: PA5
spi_software_miso_pin: PA6
spi_software_mosi_pin: PA7
axes_map: x,z,y
```

Tomadoggy, 2023-03-07:
I am just now trying to install my adxl345 on my Manta M8P / CB1 and am unable to boot with the sensor attached to SPI1.  Interestingly enough, if I hot plug the sensor into the SPI then everything works.  I can successfully execute ACCELEROMETER_QUERY, AND MEASURE_AXES_NOISE.  However, if I leave the sensor plugged in, I can not boot.  Here is my setup in printer.cfg:

Marck80, 2023-03-14:
you solved? I have the same problem

Tomadoggy, 2023-03-14:
No resolution yet...was working with a BTT tech on their FB groop and tried
a lot of things, now that tech has gone silent.
*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-
        *John*

Cambridge66, 2023-03-23:
I have the exaxt same issue M8P v1.1 - any board restart will force to re-flash firmware on MCO when the ADXL345 is connected. If I disconnect it, board boots.

Tomadoggy, 2023-03-23:
So we have it posted here...but I am not too hopeful that BTT will pick it up and respond.  I have gone to several Discord chanels and the BTT Facebook site but no resolution.  Any Ideas from you?

Marck80, 2023-03-23:
i bought "BIGTREETECH BTT ADXL345 V1.0" and with that it works.

Tomadoggy, 2023-03-27:
@RickoT I have the same ADXL, Just to verify you are not using SPI1, but connecting to PC4 and also want to make sure you can boot with the ADXL connected.  On SIP1 I can use the ADXL but only if I plug it in after booting the M8P...you do have the M8P here correct?  Hope this works, but wanted to verify with you prior to rewiring.  Thank you.

Tomadoggy, 2023-03-27:
[image]

RickoT, 2023-03-27:
[image]

RickoT, 2023-03-27:
Oops, SCA === SDA

Tomadoggy, 2023-03-27:
@RickoT  I see the disjoin now...you have version 1.1 of the board, I have version 1.0.  Same wiring different pin assignments.  Glad yours is working maybe this is an issue with V1.0 of the board.
[image]

RickoT, 2023-03-27:
Try swapping your PA6 and PA7 pins, thats what I did when it was not booting the MCU

Also, while im here, im trying to find out if I should keep the acetometer on all the time or just while tuning?  Also, if there is a purpose for leaving the accelerometer on all the time is there a way to have both axis hooked up at the same time?

Tomadoggy, 2023-03-27:
Just a very small current draw while hooked up.  NP to leave it on.  

I swapped PA6 and PA7, booted, but no communication with the ADXL.  Again, mine works just fine, but no board boot when connected.  Thus my pins are correct, but thanks for the suggestion

Flavourdynamics, 2023-04-13:
I'm also experiencing this issue. Wont boot when connected.

TFLRichard, 2023-04-17:
Wow, I am having this same issue with the Manta M8P and the ADXL345. I have noticed that I can't use the GND/+5v I had to go to the 3.3v pin because I fried my first ADXL345. So here is my issue. I can connect to the ADXL345 and I can run the accelerometer_query with results: accelerometer values (x, y, z): -9548.656652, 2298.286494, 0.000000.
My .conf:
########################################
# ADXL
########################################

[adxl345]
cs_pin: PB15
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
spi_speed: 5000000
rate: 3200
axes_map: x,z,y

#[resonance_tester]
#accel_chip: adxl345
#probe_points: 150,150,100
As soon as I uncomment [resonance_test] my printer locks up and will not boot into the firmware.
I am at the end of my rope. Please help me!

Tomadoggy, 2023-04-17:
I upgraded to the V1.1 version of the board and now can boot with the ADXL connect.  Try as I did to get it working with the V1.0 and even working directly with BTT, never achieved success.
[image]

TFLRichard, 2023-04-17:
> I upgraded to the V1.1 version of the board and now can boot with the ADXL connected. Try as I did to get it working with the V1.0 and even working directly with BTT, never achieved success. [image]

I did get mine working last night on the V1.0. What I had to do is place the ADXL section under the spot in the printer.cfg where is says BL-Touch. Once I rebooted and power cycled the printer it worked just fine.

Flavourdynamics, 2023-04-18:
> I upgraded to the V1.1 version of the board and now can boot with the ADXL connected. Try as I did to get it working with the V1.0 and even working directly with BTT, never achieved success. [image]

What do you think the solution here was? Your pin def is slightly different than mine, but you've also specified the spi speed, rate, axes map, and included the resonance tester section below. I'll give this a shot tonight.

chaaalyy, 2023-04-23:
It seems, the Manta M8P isn´t the only board with exact that problem: With the new BTT Octopus Max EZ exactly the same. Thanks for pointing to the miso/mosi swap. That did the trick for me (at least as a quick and dirty workaround). Unfortunately i can´t leave it like that, because the steppers (in case of 5160´s or similar) use the same cpu pins (shared spi) and in a couple of weeks i get other drivers / motors... I didn´t buy BTT for a long time (just problems with their stuff) and it seems, the decision to come back for the "top of the line" board also was nothing than a mistake :(

egypro98, 2023-05-04:
> I did get mine working last night on the V1.0. What I had to do is place the ADXL section under the spot in the printer.cfg where is says BL-Touch. Once I rebooted and power cycled the printer it [...]

--
Can you tell us more details please, type of adxl345 you are using, pins of adxl, etc
what i tried to do to  get mine working and escape from booting was
1- connect adxl after booting but gives an error when trying to use [resonance_tester]
2- using a rpi pico but klipper gives >> adxl345: spi pins must be on same mcu
[image]

3- using usb-adx345 provided by triangle labs but gives the same error
4- changing pins to another pin such exp pins but it gives that adxl345 is not connected
Note : i am using RATos 
Is buying an adxl345 provided by btt is the only way to fix that ?

GitRDone2022, 2023-05-13:
Same problem on my M8P v1.0  Rewired to use the other SPI channel for the screen.

Moved the pins to EXP2, (Power & GND from EXP1) and works perfectly.

[adxl345]
cs_pin: PB12
spi_software_sclk_pin: PB13
spi_software_mosi_pin: PB14
spi_software_miso_pin: PB11
spi_speed: 5000000
rate: 3200

Passed ACCELEROMETER_QUERY and TEST_RESONANCES

yuriy-panov, 2023-05-31:
Cheap chinese adxl345. Manta m8p v1.1. Did exactly as [RickoT] wrote. All works fine. No issues at all

fabio-s-franco, 2023-06-28:
> -- Can you tell us more details please, type of adxl345 you are using, pins of adxl, etc what i tried to do to get mine working and escape from booting was 1- connect adxl after booting but gives an [...]

It all depends on your setup. My struggle with it seems to be some sort of SPI channel conflict. Because both my TFT screen and MAX driver work via SPI, something is happening there that prevents proper setup od ADXL.

I still have to try connecting it without the screen and with a regular thermistor that won't require the max driver then I can also switch off the SPI bus used by the TFT in CB1 /boot/BoardEnv.txt

The fact is this board definitely has issues and if BTT was active here, perhaps not so many people would need to contact them directly which takes weeks to get a response from, which also makes diagnostics a whole lot more difficult with so much time in betwwen.

firegarch, 2023-08-04:
I am kind of stuck at the same point, trying to connect a M8P/ CB1 to the proposed ADXL345 shown in the BTT M8P manual
[image]

No luck with anything so far. Since I am running TMC5610, I am already relying on spi_software pins for all of the driver controls.

## What I have tried:
- Connecting ADXL345 to M8P as described in manual with CS_pin = PC4 and PB15
> [adxl345]
> cs_pin: PC4
> spi_software_sclk_pin: PA5
> spi_software_mosi_pin: PA7
> spi_software_miso_pin: PA6

and

> [adxl345]
> cs_pin: PB15
> spi_software_sclk_pin: PA5
> spi_software_mosi_pin: PA7
> spi_software_miso_pin: PA6

- Flipping MISO and MOSI pins in HW, just in case. --> No difference
- Unplugging the screen and using EXP2 pins 
>[adxl345]
> cs_pin: PB12
> spi_software_sclk_pin: PB13
> spi_software_mosi_pin: PB14
> spi_software_miso_pin: PB11

No matter what I do, I always get the same error message:
`Error on 'ACCELEROMETER_QUERY': missing CHIP`

The only difference I see is, that in the first configuration version mentioned (with cs_pin as  `PC4`), klipper doesn't boot with the adxl345 connected. 
I even tried a 2nd accelerometer in the form of this one: GY-291
[image]

It doesn't matter exact same behavior. I have absolutely no idea what the problem is. Any advice on how to proceed?

IVOMTB, 2023-09-26:
Hi dudes i got it!!! Vcc to 3.3 v try

#[adxl 345##################################################
[adxl345]
cs_pin: PB15
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
spi_speed: 5000000
rate: 3200
axes_map: x,z,y

[resonance_tester]
accel_chip: adxl345
probe_points: 150, 150, 20  # an example

yshui, 2023-10-04:
I am having the same problem. M8P won't boot if adxl345 is plugged in (it also seems to die if I change adxl345 config and restart klipper). But after M8P has booted, adxl345 works just fine.

I am using `cs_pin: PC4` and hardware spi.

IVOMTB, 2023-10-07:
i made this steps:
reboot M8P without connect (sdo/sda/scl) 
connect cable on board ADXL and  try ACCELEROMETER_QUERY  If you see accelerometer_value try next step...

fabio-s-franco, 2023-10-28:
> I am having the same problem. M8P won't boot if adxl345 is plugged in (it also seems to die if I change adxl345 config and restart klipper). But after M8P has booted, adxl345 works just fine.
> 
> I am using `cs_pin: PC4` and hardware spi.

I am yet to find a hardware spi that works in this board. Same problem with MAX31865 v2 (from BTT even). Just go with software (check the manual again, seems it changed since I saw it last time, uses pin PB15, instead of PC4 like before).

Flavourdynamics, 2023-12-05:
It seems like I finally got it following mostly Ricko's instructions:

Manta V1.1
Generic ADXL

[adxl345]
cs_pin: PC4
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA6 #SD0 --> MOSI
spi_software_miso_pin: PA7 #SDA --> MISO
spi_speed: 5000000
rate: 3200
axes_map: x,z,y

Notably, using opposite pins than what the Manta manual specifies. I'm using 3.3V, as the chip was quite hot on 5V. Further testing will be required to confirm the fix.

Flavourdynamics, 2023-12-06:
This change is under review. Do not apply yet.

Twisted-FPV, 2024-04-14:
I got it working on the Manta M8P v2.0 finally. 
hope this helps someone thats been searching for info on how to do it on the v2.0
[image]
[adxl345]
cs_pin: PA15 
spi_software_sclk_pin: PC10
spi_software_mosi_pin: PC12
spi_software_miso_pin: PC11
spi_speed: 5000000
rate: 3200
axes_map: x,z,y

Genocrush, 2024-09-21:
> I got it working on the Manta M8P v2.0 finally. hope this helps someone thats been searching for info on how to do it on the v2.0 ![manta SPI Pinout](https://private-user-images.githubusercontent.com/160196357/322281093-27c8bc0a-dfa5-49df-ac4d-c6e5892edb7d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTQyNTgsIm5iZiI6MTcyNjk1Mzk1OCwicGF0aCI6Ii8xNjAxOTYzNTcvMzIyMjgxMDkzLTI3YzhiYzBhLWRmYTUtNDlkZi1hYzRkLWM2ZTU4OTJlZGI3ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMTI1NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wOWU1YWYxMWQxMDRiMmRkYTM4ZmY5NDRjODMwZDk4NTNiZmRmYjAzMDFlN2NkNDhlNzZjOTA3MWFkNWQzNTdlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.JZoEuQHIrkpSh_6iqxSRKT4v5Bx4jffH-KpokkztxFA) ![20240414_020321](https://private-user-images.githubusercontent.com/160196357/322281304-d1ca2dca-1f7f-4fb8-9f9a-c15ab18f458e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTQyNTgsIm5iZiI6MTcyNjk1Mzk1OCwicGF0aCI6Ii8xNjAxOTYzNTcvMzIyMjgxMzA0LWQxY2EyZGNhLTFmN2YtNGZiOC05ZjlhLWMxNWFiMThmNDU4ZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMTI1NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00Mjg2YWViMmE4YzNkZDE4NzIxZDVmOWE5NzZjNzgzNjA3Y2M3ZjJiZDMxYzEzM2I0MWY1ZmQzM2QzZWE1YjVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.NCHFJeaKDoP4us-nAnKMrixl4wHA5QONVFJvxwZztO0) ![20240414_020353](https://private-user-images.githubusercontent.com/160196357/322281311-6fb1e1c0-15dd-417c-9cac-f36a552ef453.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTQyNTgsIm5iZiI6MTcyNjk1Mzk1OCwicGF0aCI6Ii8xNjAxOTYzNTcvMzIyMjgxMzExLTZmYjFlMWMwLTE1ZGQtNDE3Yy05Y2FjLWYzNmE1NTJlZjQ1My5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMTI1NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NmJmNzViOWIzZDcyNDkwMTNjNGM0MmEyNWMwMjIzZDE3MmQ5OGI0YzY3NDY1NzkyMTUyMzYwZjRmOTVmMDA5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.4B4HjhMG7x5mc6AAsQUwEk1DK2TQWBjprIAErwRHD0k) [adxl345] cs_pin: PA15 spi_software_sclk_pin: PC10 spi_software_mosi_pin: PC12 spi_software_miso_pin: PC11 spi_speed: 5000000 rate: 3200 axes_map: x,z,y

thank you so much i have been searching like crazy. worked perfectly without any problems. a very good guide that is very easy to follow.

Hubitski, 2026-08-20:
I’ve encountered the common bootloop issue on the Manta M8P V1.0 when connecting an ADXL345 to the SPI1 header. 

As many of you know, the board fails to initialize when the sensor is wired "by the book".

The bootloop occurs because the SD card (used for firmware/config) and the SPI1 header share the same bus. During startup, the board's bootloader attempts to poll the SD card. The ADXL345 on the same bus intercepts these signals, starts spitting errors/garbage back into the bus, which causes the SD card read to fail. This leads to an endless reboot cycle. This is a hardware issue and cant be fixed by a firmware update.

The Fix: The workaround is to bypass the hardware SPI initialization at boot by using software SPI and swapping the MOSI/MISO pins on ADXL and in the config.

Working Config:

[adxl345]
cs_pin: PB15
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA6
spi_software_miso_pin: PA7

After applying these changes, the board boots normally and the sensor is fully functional: ACCELEROMETER_QUERY: accelerometer values (x, y, z): 444.123565, 0.000000, 10648.727422