bigtreetech/Manta-M8P #28: screw size for CB1/CM4

jscancella, 2023-01-05:
What screw size should be used for the CB1/CM4 mount? I tried 2mm and 2.5mm but neither seemed to want to screw in.

jfneptune, 2023-01-06:
The boards just snap in place.  It takes a little more force than I care for, but it will just snap in place.  No screws are needed.

jscancella, 2023-01-07:
> The boards just snap in place. It takes a little more force than I care for, but it will just snap in place. No screws are needed.

Yes I did that, but I saw that there are 4 mounting holes for screws and I wanted to use those as well.

thisiskeithb, 2023-01-07:
M2.5, but there is tape covering the standoffs that will need to be removed.

anscarlett, 2023-07-16:
Strange that it isn't written on the dimensional drawings anywhere. Disappointing that the screws weren't supplied with the board.

jpereira-ca, 2023-07-16:
> M2.5, but there is tape covering the standoffs that will need to be removed.

Yes, there are tiny little round translucent orange stickers protecting the screw holes. After removing them, the M2.5 screws fit perfectly.

anscarlett, 2023-07-18:
For reference, I found a review on amazon saying that with the heatsink, you need 2x M2.5x8mm and 2x m2.5x6mm screws. Would have been nice to have designed the heatsink so only one size screw was needed.