bigtreetech/Manta-M8P #30: CanBus support

emersonsc, 2023-01-11:
The owners manuals provided give zero help on getting the built in canbus to work. Theres no pin outs either.

Vargsh0rd3, 2023-01-17:
i agree. Im so close to just sledge hammering shit. i feel like ive watched and read 1000 bits a piece, and im still no closer. and fairly certain ive bricked an unnecessary U2C

emersonsc, 2023-01-17:
> i agree. Im so close to just sledge hammering shit. i feel like ive watched and read 1000 bits a piece, and im still no closer. and fairly certain ive bricked an unnecessary U2C

Yeah its annoying how spread out and fragmented documentation is for CanBus. For as much of a hard@$$ O'Conner is about stuff, you'd think he wouldnt merge something like CanBus into Klipper without proper documentation...

Vargsh0rd3, 2023-01-17:
Ingot my voron at the end of September. Had it running  foe maybe 2 weeks. Gotna few mods and lost a month. Finally got it running for another week. Tap arrived. Fried my board and all my LEDs because of poor documentation on a necessary alteration to hartk pcb. 
Dfh took 4 weeks to send me 600 bucks of stuff that I now can't get working.
I'm starting to feel really stupid.

emersonsc, 2023-01-17:
> Ingot my voron at the end of September. Had it running foe maybe 2 weeks. Gotna few mods and lost a month. Finally got it running for another week. Tap arrived. Fried my board and all my LEDs [...]

what mods? are you on discord we can  move this conversation there and I could possibly help

Vargsh0rd3, 2023-01-17:
No those were like 2 crisis ago lol. I gotnthat stuff figured now. 
We have the same problem now. And i dunno. I know not probably help matters. But I don't feel like I learned anything if I get help. I needn't scrape and struggle and curs4nthe sky and smash a 20 year old monitor with a sledgehammer that I keep around for therapy rage.

Vargsh0rd3, 2023-01-17:
Healthy eh?

Vargsh0rd3, 2023-01-17:
Holy sausage thumbs. I need to proofread. Too much repetitive mumbo-jumbo. Too lazy. Eyes... searing shut.

ALM007cs, 2023-02-09:
This is a feature that is also advertised on the homepage and in the sales documents, but there is no documentation on how to use it.
How is the Can-Bus functionality activated when using a CB1?
Not a word about it in any documentation from Biqu or BTT.

emersonsc, 2023-02-12:
> This is a feature that is also advertised on the homepage and in the sales documents, but there is no documentation on how to use it. How is the Can-Bus functionality activated when using a CB1? Not a word about it in any documentation from Biqu or BTT.

This is something Biqu/Bigtreetech does with pretty much everything they sell. They make lots of claims on the productpages, then when it comes to support its crickets except for the cheerleaders who get upset when you call our Biqu/BTT for this kind of crap.

alexwiecke, 2023-02-14:
Hey guys, this is the procedure we follow for getting CANBUS working on the Manta v1.1s:

These instructions reference the Manta M8P User Manual v1.04 (Commit 53df01d on Jan 8)

1) Flash the compute board (CB1 or CM4) following the normal procedure in section 4 or 5 of the manual
2) SSH into the compute board as shown in section 6.1
3) In the terminal, navigate to the klipper repository and open the build configuration:
``` bash
cd ~/klipper
make menuconfig
```
4) Follow the compile instructions shown in section 6.2, however compile with the configuration shown below in the first step:
``` menuconfig
[\*] Enable extra low level configuration options
	Micro controller Architecture (STMicroelectronics STM32 )
	Processor model (STM32G0B1 )
	Bootloader offset 8 KiB bootloader)
	Clock Reference (8 MHz crystal)
	Communication interface (USB to CAN bus bridge (USB on PA11/PA12))
	CAN bus interface (CAN bus (on PD12/PD13))
(250000) CAN bus speed
```
[image] 

5) Press 'q' to exit, save the configuration
6) Build the firmware:
``` bash
make clean
make menuconfig
```
7) Put the Manta MCU in bootloader mode:
	1) Hold down the BOOT0 button
	2) Press and release the RESET button
	3) Release the BOOT0 button
8) Flash the MCU:
``` bash
make flash FLASH_DEVICE=0483:df11
```
9) Look for the success message. If the file does not download successfully, redo step 7 and try again. 

[image]

10) Configure the CANBUS UUID for the manta
Get the MCU's UUID
``` bash
~/klipper$ ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
```
Expected response:
``` bash
> Found canbus_uuid=[UUID], Application: Klipper
> Total 1 uuids found
```

If the UUID doesn't appear, make sure you've setup the host as recommended in the instructions under [Host Hardware](https://www.klipper3d.org/CANBUS.html#host-hardware), then reflash the board per steps 7-9

11) Wire up the CAN bus using a twisted pair between the manta and your other boards. 
See the Manta M8P v1.1 pinout on page 11 of the manual. 
Make sure to verify that you have 120ohm resistors terminating the bus, and that your other boards are flashed for CANBUS communication at the same speed (250000)

**Once you've gotten to this point, the [Klipper CANBUS documentation](https://www.klipper3d.org/CANBUS.html) will get you the rest of the way**

ALM007cs, 2023-02-14:
@alexwiecke It Works.... THX

Only problemv i got was the bootloader. I have to flash the firmware by sd -card.

Now CAN-bus communication to EBB36 work well - without additional boards.

alexwiecke, 2023-02-14:
Hey guys, this is the procedure we follow for getting CANBUS working on the Manta v1.1s:

These instructions reference the Manta M8P User Manual v1.04 (Commit 53df01d on Jan 8)

1) Flash the compute board (CB1 or CM4) following the normal procedure in section 4 or 5 of the manual
2) SSH into the compute board as shown in section 6.1
3) In the terminal, navigate to the klipper repository and open the build configuration:
``` bash
cd ~/klipper
make menuconfig
```
4) Setup the configuration shown below:
``` menuconfig
[\*] Enable extra low level configuration options
	Micro controller Architecture (STMicroelectronics STM32 )
	Processor model (STM32G0B1 )
	Bootloader offset 8 KiB bootloader)
	Clock Reference (8 MHz crystal)
	Communication interface (USB to CAN bus bridge (USB on PA11/PA12))
	CAN bus interface (CAN bus (on PD12/PD13))
(250000) CAN bus speed
```
[image] 

5) Press 'q' to exit, save the configuration
6) Build the firmware:
``` bash
make clean
make
```
7) Put the Manta MCU in bootloader mode:
	1) Hold down the BOOT0 button
	2) Press and release the RESET button
	3) Release the BOOT0 button
8) Flash the MCU:
``` bash
make flash FLASH_DEVICE=0483:df11
```
9) Look for the success message. If the file does not download successfully, redo step 7 and try again. 

[image]

10) Check that the MCU is showing up as a CANBUS device:
``` bash
~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
```
Expected response:
``` bash
> Found canbus_uuid=[UUID], Application: Klipper
> Total 1 uuids found
```

If the UUID doesn't appear, make sure you've setup the host as recommended in the instructions under [Host Hardware](https://www.klipper3d.org/CANBUS.html#host-hardware), then reflash the board per steps 7-9

11) Wire up the CAN bus using a twisted pair between the manta and your other boards. 
See the Manta M8P v1.1 pinout on page 11 of the manual. 
Make sure to verify that you have 120ohm resistors terminating the bus, and that your other boards are flashed for CANBUS communication at the same speed (250000)

**Once you've gotten to this point, the [Klipper CANBUS documentation](https://www.klipper3d.org/CANBUS.html) will get you the rest of the way**

danielsmoura, 2023-02-16:
This work, but how i get the serial of Manta, because when i try run "ls /dev/serial/by-id/" the command return to me this error:
ls: cannot access '/dev/serial/by-id/': No such file or directory

Alex-Pantheon, 2023-02-16:
> when i try run "ls /dev/serial/by-id/" the command return to me this error: ls: cannot access '/dev/serial/by-id/': No such file or directory

Once flashed in [CAN bridge mode](https://www.klipper3d.org/CANBUS.html#usb-to-can-bus-bridge-mode) the MCU no longer appears as a serial device, but instead shows up as a CAN device on `can0`. So you'll need a CAN UUID, not a Serial ID.

See the [Klipper CANBUS documentation](https://www.klipper3d.org/CANBUS.html) for instructions on how to query and configure it.

danielsmoura, 2023-02-16:
Im not need put on printer.cfg the ?:

[mcu]
serial: /dev/serial/by-id/xxxx

What i put here ?

Alex-Pantheon, 2023-02-16:
@danielsmoura, instead of a serial address, you'd put a CAN UUID, see below:
```
[mcu]
#serial: not needed
canbus_uuid: 6cc687f42f48
```

ALM007cs, 2023-02-16:
@danielsmoura :
https://www.klipper3d.org/CANBUS.html

twingi0815, 2023-02-28:
every time I turn the printer off and on again it shows me

~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0

 Total 0 uuids found

when i flash the board new is show 1 uuid .

when i turn the printer off and on it shows total uuids 0

EveCodeJunkie, 2023-03-04:
> every time I turn the printer off and on again it shows me
> 
> ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
> 
> Total 0 uuids found
> 
> when i flash the board new is show 1 uuid .
> 
> when i turn the printer off and on it shows total uuids 0

Once you update your printer.cfg with the canbus_uuid and restart, klipper will claim the device.  Running canbus_query.py will only show new or otherwise unused canbus devices.

twingi0815, 2023-03-05:
> Once you update your printer.cfg with the canbus_uuid and restart, klipper will claim the device. Running canbus_query.py will only show new or otherwise unused canbus devices.

after a restart, my board in klipper also disappeared

twingi0815, 2023-03-05:
no it disappears even if i don't write the uuid in klipper

that's life shit like the flashing of the ebb42

Mickd02, 2023-03-06:
Vargsh0rd3 and emersonsc i feel your pain been at it 3 weeks and no luck and pure frustration. my setup is pure BTT Manta m8P v1 ,CB1 and Io2Can, originally started with SB2240 but being new to this whole tmc2240 thing NOT supported natively!!  so thought give it a go and ordered a SB2209 seeing as it looks to be supported, thinking should be easy as now got all the products that "should work together" BUT hard NO!! really think BTT rushed this out before Holidays hoping some would work it out for them. BTT need to realize that not everyone is a programming genius and need to write decent instructions that make sense and not 3 different ways to do the same task as it all melds into??? a product that doesn't work as advertised.
I also see the Github Klipper Branches for TMC2240 are no good also .. machine now going back to hard wiring and deal with the can bus in 6 months when it all gets sorted. 
[image]

Jjustice1993, 2023-03-12:
So I have been working on this throughout the morning, I have got to the point where i can connect to the MCU board (Manta) via CANbus, but once i plug the hermit crab in it crashes the CAN network, following schematics from BTT GitHub for each the Manta m8p and hermit crab tool board i have made a connector, once this happens even after a hard reboot and disconnecting the hermit crab from the board the CAN network will not recover. Any ideas? Do i need to define the CAN pins in the klipper config?

patrickdiesslin, 2023-03-25:
Thanks @Alex-Pantheon could I please ask for some help. 
Where you wrote; 
_Wire up the CAN bus using a twisted pair between the manta and your other boards.
See the Manta M8P v1.1 pinout on page 11 of the manual._

Could you please show which pins on that page to use for the canbus connection and the pinout of those pins please. 
Thanks in advance

EveCodeJunkie, 2023-03-25:
To use the built in CAN Bus on the Manta M8P, you need to use pins PD12/PD13.

omorgan831, 2023-04-16:
Just want to make sure I understand the physical connection side of stuff... the firmware bit makes sense.

When using a CM4/CB1 with the Manta v1.1 is the can comms done through the B2B connectors for the CM4/CB1 or does it need the likes of a Pi Can hat?

I ask as I'm building / looking to build a system with a preexisting hermit crab can toolhead (the hermit crab comes with a pi hat, and a tool board). 

I 'assume' the PD12/13 connectors on the V1.1 board would go to the hermit crabs tool board (a 'dumb' hub) but does it still need the pi hat to connect to the tool board as well?

Can the v1.1 use Canboot or can it be re-flased via the klipper tools as described above after the MCU is flashed the first time (assume again you'd point your binary to can0 uuid). 

Any advice would be appreciated. Whilst I HATE BTT for their overall lack of support once schematics and rough manuals have been uploaded (very much heres the hardware work it out yourself), they do make hardware that fits with what I want to achieve at a price point that makes sesne...... and I know I'm lucky enough to be able to decipher (more or less) the info they do upload.

EveCodeJunkie, 2023-04-16:
> Just want to make sure I understand the physical connection side of stuff... the firmware bit makes sense.
> 
> When using a CM4/CB1 with the Manta v1.1 is the can comms done through the B2B connectors for the CM4/CB1 or does it need the likes of a Pi Can hat?

The CAN and the rest of the MCU communicate with the CM4/CB1 through the B2B connectors.  There is no need for a seperate CAN hat.  You can use a separate CAN hat if you want to offload the CAN communications and not use the built in CAN.  There is plenty of CPU and bandwidth on the Manta so I don;t see this as usefull.

> I ask as I'm building / looking to build a system with a preexisting hermit crab can toolhead (the hermit crab comes with a pi hat, and a tool board).
>
> I 'assume' the PD12/13 connectors on the V1.1 [...]

If you are using the CAN version of the Hermit Crab Tool board, I see no reason why you would need the Pi CAN Hat.  

> Can the v1.1 use Canboot or can it be re-flased via the klipper tools as described above after the MCU is flashed the first time (assume again you'd point your binary to can0 uuid).

The 1.1 can use CanBoot, but I see no reason to.  The Manta is directly connected to the CM4/CB1 and always will be.  So flashing/reflashing is quite easy and direct.  You would only need CanBoot if you want to flash devices that are only connected to the CANBus.

> Any advice would be appreciated. Whilst I HATE BTT for their overall lack of support once schematics and rough manuals have been uploaded (very much heres the hardware work it out yourself), they do [...]

Yeah, the lack of support sucks.  Took me forever to get my Manta MP8 v1.1 w/CB1  and EBB SB2209 working together.  Three of their own products.  You would think that there would be a step by step guide somewhere, but nope.  You got to figure it all out yourself.

omorgan831, 2023-04-16:
Huge thanks,

Yh as its using the B2B then that all makes sense. And again, no need for canboot as its using that.

Seriously thanks for your time and support!!

nikolicjakov, 2023-04-24:
@EveCodeJunkie 

Hi there! Your comment was helpful in order to understand that i do not need to use Pi HAT that came with my Hermit Crab CAN.

Now i need to figure out how i could cut off one end of phone cable that came with the device and crimp connectors for my Manta M8P motherboard CAN pins.

Do you mby know where i could find pinout for BIGTREETECH CAN-Tool Board Version V1.0.

I only could find pinout for my motherboard..

[image]

But i would need to know the other end also.. :(

Rizz226, 2023-04-24:
If I'm understanding you correctly, you'll find the pinouts in this repo:

https://github.com/bigtreetech/EBB

So on one end, you'd have your manta & a positive and negative lead from your PSU. On the other side of the cable will be whichever EBB CAN board you have.

nikolicjakov, 2023-04-24:
That is correct i have Hermit Crab CAN and Manta M8P. Now i have them connected without using RPi Hat and now i need to understand how to configure software part.

Do i need to flash some custom klipper binary to Hermit Crab CAN and how could i use 2 mcu-s (Manta M8P and Hermit Crab CAN) in klipper...

Rizz226, 2023-04-24:
@nikolicjakov - for the simplest setup, you'd flash your hermit crab with whatever is normal for that particular board. For the manta however, you'll have to choose the connection type "USB TO CAN BRIDGE" when building it's firmware. Reference the pinout for the specific option to choose. After that, you'll run the can_query script from the Klipper CANBUS documentation and you should see 2 uninitialized devices. Top one was the Manta for me.

nikolicjakov, 2023-04-26:
After a day of thinkering with Crab CAN and M8P i have managed to get it working. But do not ask me how since i do not know i have compiled and flashed klipper milion times both on manta m8p and crab CAN. After some time it just booted and worked.

So i can only confirm that combinantion where you have Manta M8P and Hermit Crab CAN can work togather without RPi HAT but direct from motherboard to CAN board and then from CAN board to Hermit Crab via USB type C cable.

bthubs, 2023-05-04:
I got CAN to work on my m8p + cb1 and sb2209. For those who are still trying to figure it out using the guide, there's a very important step that is not included as detailed by Alex-Pantheon. What confused me most was flashing two different configs of klipper, one is for the SB2209, the other is for the Manta M8P MCU. I'll detail how I got the m8p to communicate with the SB2209:

1. Flash CANboot to the SB2209 (CAN PB0/PB1) (follow guide)
2. Flash klipper to SB2209 (CAN PB0/PB1) (follow guide)
3. Flash klipper to M8P MCU as detailed by Alex-Pantheon (USB to CAN bus bridge PA11/PA12 + CAN bus interface PD12/PD13)
4. Run `~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0`, this should show 2 CAN bus IDs both running klipper, if not repeat steps 1-3.
5. Under printer.cfg [mcu] section, comment out serial, add canbus_uuid: ###### (for M8P CAN bus ID)
6. Under BTT sample cfg for SB2209 [mcu EBBCan] section, comment out serial, add canbus_uuid: ###### (for SB2209 CAN bus ID)
7. Restart

Hope this helps!

Alex-Pantheon, 2023-05-04:
Yeah, every CAN board has its own flavor of flashing procedure and config, you'll have to follow the flashing instructions specific to your board.

djeZo888, 2023-05-17:
Ah thanks! So, only two wires are actually needed then.

From here: https://github.com/bigtreetech/HermitCrab/blob/master/CanbusKlipper/Hardware/Schematic/BTT%20Tool%20Board-SCH.pdf
I see that Hermit Crabs Toolboard RJ11 CAN-IN port takes in two sets of CANs: CAN0 and CAN1. Does it matter which one is connected to M8P? Looking at schematic, it makes sense to use CAN1-L and CAN1-H. CAN0 only goes from CAN-IN to CAN-OUT.

seb449, 2023-05-17:
Hi, my card communicates well, Found canbus_uuid=030792b2fc33, Application: CanBoot
but on klipper hs, with
[mcu can0]
canbus_uuid: 030792b2fc33
not ok  !!!

    allow-hotplug can0
    iface can0 can static
         bitrate 250000
         up ifconfig $IFACE txqueuelen 2048

Help

nikolicjakov, 2023-05-22:
BIGTREETECH IF YOU READ THIS YOU SHOULD RLY IMPROVE ON YOUR DOCUMENTATION!

IT IS NOT ONLY IMPORTANT TO SELL YOUR EQUIPMENT BUT TO SUPPORT YOUR CUSTOMERS.

jpereira-ca, 2023-06-23:
For all those out there wondering what the pinouts and wiring diagram between the Hermit Crab CAN Tool Board and the M8P v1.1 CAN headers is, here it is:

When you are looking at the tool board from top, the 3rd and 4th wire from the left, are CAN-H and CAN-L respectively, and when you look at the M8P Can header pins from left to right are CAN-L, and CAN-H, respectively.

albielew, 2023-07-11:
I have the Manta M8p and considering the Can version of Hermit Crab
but I see the 24V is sent through usb-c cable
Surely usb-c cable is not thick enough to handle the extruder current?

shoikan, 2023-07-22:
>  Surely usb-c cable is not thick enough to handle the extruder current?

As always 'that depends'. I had a a 'fruit-branded laptop, with an i9 CPU that got charged via a USB-C cable with a 90W charger. Heater cartridges tend to be between 30 and 70W. So that fits. Additionally, the 'PD' ('Power Delivery') spec of USB-C lists 20V as the highest end voltage (and even 48V for some extended spec, though I am assuming we won't see that 'in the wild' very often).

Pneumanifest, 2023-07-25:
> 

If you use a power delivery USBC cable it is fine. It can handle up to 100W with some newer ones being capable od 250W. Just note that not all USBC cables are power delivery cables.

DP4US, 2023-08-07:
I have an m8p v1.1 and ebb36. There is a resistor on the ebb to jumper that is documented. There is a resistor on the m8p to jumper but not documented. It is next to the can connector. I was getting dropped connection at 250k with 1048 and 500k at 2048. Tried both speeds. Once the jumper was placed on the board no more dropped connections on either speed setup. Went through a lot of configs, apt installs, crimping and testing to find this resistor afterwards. Has not faulted since and 100 print hours later.

Pneumanifest, 2023-08-07:
That's called a termination resistor and is required for canbus. You must always have termination resistors at both ends of the bus lines. 120R is the most common value used. Not having that in place will allow the signal to echo causing all sorts of connection issues.

DP4US, 2023-08-08:
> That's called a termination resistor and is required for canbus. You must always have termination resistors at both ends of the bus lines. 120R is the most common value used. Not having that in place will allow the signal to echo causing all sorts of connection issues.

The m8p does not have this documented at the time of my install. Hope this helps someone.

Pneumanifest, 2023-08-08:
> The m8p does not have this documented at the time of my install. Hope this helps someone.

It is documented in the schematic and printed on the board itself. Depending on how the CAN bus line is being set up it may or may not be needed. There are different configurations so it's not always required, but all can bus lines need the termination resistors at both ends of the bus. If the m8p is at one of the ends it needs it if it's not at the end, for example 2 ebbs, it doesn't need it. It's not anything unique to this board.

DP4US, 2023-08-13:
This must be documented in the M8P manual. The CAN is labeled on the board but newbies to CAN you will need some better documentation. Also needs the same verbiage as the EBB for CAN in the M8P CAN section. An updated picture to show the resistor and when to use it should be added. BTT please ensure this is in all versions that have the CAN bridge. Install is not clear and can be erratic if both are not terminated. Seems 50 50 so please document . All other parts of the CAN are documented well. I believe the speed can be bumped up to 500K and 2048 bit. Not having any issues at this speed [double of manual]. M8P|EBB36 CAN Bridge setup.

Dye007, 2023-08-26:
> Ich habe ein m8p v1.1 und ebb36. Es gibt einen Widerstand am Ebbe-zu-Jumper, der dokumentiert ist. Es gibt einen Widerstand am M8P-zu-Jumper, der jedoch nicht dokumentiert ist. Es befindet sich neben dem Dosenanschluss. Die Verbindung wurde bei 250.000 mit 1048 und bei 500.000 mit 2048 unterbrochen. Ich habe beide Geschwindigkeiten ausprobiert. Sobald der Jumper auf der Platine platziert wurde, gab es bei beiden Geschwindigkeitseinstellungen keine Verbindungsabbrüche mehr. Habe viele Konfigurationen, passende Installationen, Crimpen und Tests durchgeführt, um diesen Widerstand anschließend zu finden. Zunächst und 100 Druckstunden später gab es keine Mängel mehr.

Kannst du mir zeigen ,wie du das angeschlossen hast?

DP4US, 2023-08-27:
[image]
[image]
The are the locations for the jumpers. If you need more help please let me know. I had to google the ebb36 connections for power and can h/l.

JuanSebastianOtalora, 2023-09-21:
> For all those out there wondering what the pinouts and wiring diagram between the Hermit Crab CAN Tool Board and the M8P v1.1 CAN headers is, here it is:
> 
> When you are looking at the tool board from top, the 3rd and 4th wire from the left, are CAN-H and CAN-L respectively, and when you look at the M8P Can header pins from left to right are CAN-L, and CAN-H, respectively. Also note that the jumpers for the CAN bus termination resistors need to be in place on both boards (on the M8P, it's the header labeled 120R, and on the HC CAN Tool Board, it's the header on the left of the RJ11 ports).
> 
> [image]
> 
> [image]

Is that for CM4 or CB1. I was told by Biqu R&D Justin, that for Hermit Crab CAN with CB1 I would need the IO2CAN v1.0 adapter (same one used for the TFT35 SPI v2.1) to make it work with the CB1. Since that information wasnt available and seems that still not available i didnt purchase the IO2CAN V1.0 so i had to use the Hermit Crab CAN with a raspi 3 and skr e3v2
However there is a new version of the Manta M8P v2.0 that has I2C, not sure if the same as the IO2CAN v1.0 and if it works with CB1. Im still at a loss. 
Would be a good idea to invest in the M8P v2 rather than getting the v1.1 to make the hermit crab can work with CB1?

Pneumanifest, 2023-09-21:
The m8p v1.1 and likely the V2 will have no problems using the canbus out with the CB1. You put it in can bridge mode and you're good to go. The original m8P didn't have a canbus transceiver so that's probably why you were told that you needed the IO2Can adapter.