bigtreetech/Manta-M8P #32: Connecting creality sonic pad to manta m8p

Ehukai466, 2023-01-15:
Hello Guys,
I have a creality ender 5 plus with a bigtreetech manta m8p board + cb1. This seems to work as I do not get any errors on the web interface. But now I wanted to add as well the Creality Sonic Pad. But it seems that the sonic pad doesn't get connection to the manta board. I've connected the sonic pad to the USB type c connector. Could it be that this is causing the issue? Does somebody know if I need to use the USB Type A port?

Freddyb888, 2023-03-11:
Following

frutabruta, 2023-04-02:
Combining Manta with Sonic pad is not possible. The USB connectors on the Manta are only connected to the CB1/CM4, not the Manta microcontroller.

Rlinhos, 2024-02-09:
I'm trying to do the same with my Sonic Pad and I just read this. We have to wait for CB1 to arrive.