bigtreetech/Manta-M8P #33: Bug NEOPIXEL

Foose83, 2023-01-22:
Hello I am having a problem with my Manta M8P V1.
The neopixel LEDs light up partially or they light up in the wrong color or do not go out at all.
I didn't have this problem with my octopus V1.1 card.
has anyone had this problem before do you have any idea what could be wrong thank you in advance.

emersonsc, 2023-01-23:
I corrected the issue by putting a 100ohm resistor on the data line just after the board. Lots of noise ends up on the data line which causes issues, the resistor helps clean it up.

Foose83, 2023-01-27:
Merci pour votre réponse, j'ai essayé 100 ohms et 680 ohms mais cela ne fonctionne toujours pas.
Je continue mes recherches.

NegussugeN, 2023-12-20:
Did anyone ever figure this out? I'm having the same exact issue.

sajrashid, 2024-09-10:
Im having same issue on v2

sajrashid, 2024-09-17:
This is what I have tried board is Manta M8p v2 

- Inline resistor (data line) 220 and 100 ohms  😔 
- Capacitor small value on the 5v line  😔 
- 5v 6a SMPS with a shared ground 😔 

I got to light up accidently as the ground was loose, but only about half lit up and did not respond to software changes (apart from a slight flicker)

So has the v2 gone backwards in the LED game