bigtreetech/Manta-M8P #37: Problems with flashing via DFU

Seoman76, 2023-02-10:
Hi all

Tried to flash my board via DFU and followed the instructions of the manual. 

All went quite well the way as shown in the manuel  included the error warning  that could be ignored 

But my mcu Version number still shows the old version instead of the new one 

Has the firmware been flashed or not?

Seoman76, 2023-02-10:
dfu-util: Invalid DFU suffix signature dfu-util: A valid DFU suffix will be required in a future dfu-util release!!! Opening DFU capable USB device... ID 0483:df11 Run-time device DFU version 011a Claiming USB DFU Interface... Setting Alternate Setting #0 ... Determining device status: state = dfuIDLE, status = 0 dfuIDLE, continuing DFU mode device DFU version 011a Device returned transfer size 1024 DfuSe interface name: "Internal Flash " Downloading to address = 0x08002000, size = 26644 Download [=========================] 100% 26644 bytes Download done. File downloaded successfully dfu-util: Error during download get_status Failed to flash to /dev/serial/by-id/usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00: Error running dfu-util If the device is already in bootloader mode it can be flashed with the following command: make flash FLASH_DEVICE=0483:df11 OR make flash FLASH_DEVICE=1209:beba If attempting to flash via 3.3V serial, then use: make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32g0b1xx_3E0009000A504B4633373520-if00 make: *** [src/stm32/Makefile:90: flash] Error 255

manotroll, 2024-04-28:
I found the solution to the problem in a video on YouTube, I'll put it here so you can see
There is an error in the PDF explanation in the manual on how to do it
https://www.youtube.com/watch?v=k8DmTrY2GgY