bigtreetech/Manta-M8P #40: Retract goes the same direction as extrude

mbruttig, 2023-02-20:
Hi, 

I'm running a modified X5SA with an LGX large on a Manta M8P and for some reason my extruder retraction goes the same direction as extrude.  I've tried different motor ports, TMC2209 drivers, and plugging in a standard NEMA17 motor but when I do a G1 E-10 F300 the filament extrudes and does not retract.  Note, the "!" on dir_pin does not change the direction for either extrude or retract.  Here's my extruder config:

[extruder]
#Motor 6
step_pin: PA10
dir_pin: !PA14
enable_pin: !PA15
microsteps: 16
rotation_distance: 8
nozzle_diameter: 0.4
filament_diameter: 1.75
heater_pin: PE3 # HE0
sensor_pin: PA1 # T0
sensor_type: Generic 3950
min_temp: 0
max_temp: 450

[tmc2209 extruder]
uart_pin: PF8 
run_current: 0.800
stealthchop_threshold: 999999

I've also tried without the TMC2209 settings and same issue. 

Cheers

RandomUser2030, 2023-02-27:
Hi - I'm stuck and it's driving me nuts.   I have replaced multiple cables and motors, and change controller ports and dir_pin settings.  No matter what I do, the extract and retract always turn clockwise - both in always the same direction.  I've emailed BIQU but they haven't been able to resolve the issue yet.  Has anyone seen this before?  Thx

RezezRX79, 2023-03-09:
I know how you feel I am having the same issue with the manta M8p 1.1 card. I have been looking in to it and it and if you look at the board schematic it appears that stepper 6 is using a different component for its direction pin than all the other steppers. for now I just swapped the servo I was going to use on 6 to 8. If anyone can figure out the right pin call out (AKA PA14, PA15, PD15 or what ever) that would be great.
[image]
[image]

12voltnick, 2023-03-15:
I think I figured it out! when I was doing the config for the stepper motors I was looking at a V1.0 pinout, but i have a V1.1 board. the pinouts are similar but not the same! and some of the pins are used for other things. Check your pinout version against your board version and this should give you the correct pin labels for step, direction, enable, and uart pins.

RandomUser2030, 2023-03-15:
Hi 12voltnick - that was the problem.  I have a v1.0 board and all the pinouts are the same for both boards up to motor6, so I didn't make the connection.  I'm surprised BTT didn't make the connection.  I had put the project on hold until I read your post.  Thank you!