bigtreetech/Manta-M8P #42: Manta M8P + CM4 , no usb connection , not flashing from SD card

Erik-Erikson, 2023-02-22:
I have just installed a CM4 onto a (V1.0) M8P

I can SSH into the board just fine and have compiled the firmware.
When I use a SD card to first flash the firmware-usb.bin (renamed to firmware.bin) nothing happens.
I tried renaming the firmware-usb.bin to FIRMWARE.bin to no effect.

Further more when using the command : ls /dev/serial/by-id/* to list the MCU serial the responce is : 
No such file or directory.

I have enabled the USB2.0 HUB in the config.txt
To check usb functionality I connected a SKR1.4 to the USB-A connection and running the command again it does find it
So the config.txt is edited Ok i think.
[image]

looxonline, 2023-02-26:
You may need to use the SWD interface to program the MCU.

looxonline, 2023-02-26:
Also, check that your OS is not dumping any hidden files onto your SD card. macOS does this sometimes and it messes with the bootloader.

QuinFromCanada, 2023-02-26:
Check what file system your SD card is formatted for.  I used a new SD card out of the box, and I didn't check and had the same issue.  after I formatted it to FAT32 it worked.

PSpiel, 2023-03-20:
Hello, Similar problem... though I had to find an old and smaller SD card (2GB) as my 32 GB new SD would not work.
Hope that helps others too.

Alexpanai, 2023-05-09:
I had the same issue. Defiantly a smaller card (not 64Gb as I tried for 2 days) formatted in FAT32 was the way to go. Strangely enough I didn't manage to format in FAT32 the bigger cards.

alex-crr, 2024-01-16:
I have a 2Gb FAT32 sd card, and it is still not working. When trying through DFU i have no serial Id created so I assume it is not working either. I have another board, MantaE3eZ for which it worked perfectly fine. Any more insights on how to solve?

manotroll, 2024-04-28:
I found the solution to the problem in a video on YouTube, I'll put it here so you can see
There is an error in the PDF explanation in the manual on how to do it
https://www.youtube.com/watch?v=k8DmTrY2GgY

DaveuRrr, 2025-03-30:
I am not sure if you are intending to use canbus but I was able to get it to work when flashing klipper at `0x08002000` address. I was using these instructions and as able to get my usb connection back `/dev/serial/by-id/`

[Manta M8P & BTT EBB SB22XX](https://forum.vorondesign.com/threads/manta-m8p-btt-ebb-sb22xx-setup.1697/)