bigtreetech/Manta-M8P #46: Manta M9P 1.1 Motor 6 Direction?

RezezRX79, 2023-03-09:
I am having an issue with the manta M8p 1.1 card. I have been looking in to it and it and if you look at the board schematic it appears that stepper 6 is using a different component for its direction pin than all the other steppers. for now I just swapped the servo I was going to use on 6 to 8. If anyone can figure out the right pin call out (AKA PA14, PA15, PD15 or what ever) that would be great. It looks like M6DIR goes threw to SWCLK and that goes to pin PA14 But PA14 doesn't change the direction of the motor.
[image]
[image]

azhago, 2023-03-13:
Same issue here.
Sending a STEPPER_BUZZ stepper="the stepper on motor#6" : it goes only one way instead of changing direction at each buzz

12voltnick, 2023-03-14:
Same issue here! Just built a voron and have been banging my head against the wall trying to figure this out as m6 is my z3 stepper driver. WTF BTT this is bs.

12voltnick, 2023-03-15:
I think I figured it out! when I was doing the config for the stepper motors I was looking at a V1.0 pinout, but i have a V1.1 board. the pinouts are similar but not the same! and some of the pins are used for other things. Check your pinout version against your board version and this should give you the correct pin labels for step, direction, enable, and uart pins.

azhago, 2023-03-15:
Triple checked, i have 1.1 board and use 1.1 pinout.

sergelhr, 2023-03-23:
ran into same issue, klipper voron cfg error-->dir_pin: !PD15 instead !PA14 wtf

NAPCAL, 2023-04-01:
@sergelhr I just posted a pull request with a correct v1.1 Klipper cfg file.
[54](https://github.com/bigtreetech/Manta-M8P/pull/54)

Synapsis181, 2023-04-24:
I'm having the problem with Motor 8  using the  pin out I keep getting an error when i try a Stepper_buzz.
Can someone tell me the setting they used. Using with with RatOS.  Its not the first time, other had reported the same problem.

NAPCAL, 2023-04-24:
Is this what you have?

# Motor8
#[extruder3]
#step_pin: PD8
#dir_pin: PC6
#enable_pin: !PC7
#heater_pin: PE1 # HE3
#sensor_pin: PA4 # T3

Synapsis181, 2023-04-24:
Not at home right now have to check, i'm using it wth Rats as stepper Z2.

NAPCAL, 2023-04-24:
The v1.1 has different pin assignments for motors 6-8.

My sample M8P v1.1 cfg has been merged into BTT GitHub, you can find the correct pin assignments there.

Synapsis181, 2023-04-24:
I also have the V1.1, thanks i'll go take a look

Synapsis181, 2023-04-24:
Still cann't get it to work.  I have this as the base config
aliases:
# steppers
  x_step_pin=PE2,    x_dir_pin=PB4,    x_enable_pin=PC11,   x_uart_pin=PC10,   x_diag_pin=PF3,    x_endstop_pin=PF3,
  y_step_pin=PF12,   y_dir_pin=PF11,   y_enable_pin=PB3,    y_uart_pin=PF13,   y_diag_pin=PF4,    y_endstop_pin=PF4,
  z0_step_pin=PA10,  z0_dir_pin=PA14,  z0_enable_pin=PA15,  z0_uart_pin=PF8,   z0_diag_pin=null,
  z1_step_pin=PD11,  z1_dir_pin=PD9,  z1_enable_pin=PD15,  z1_uart_pin=PD14,  z1_diag_pin=null,
  z2_step_pin=PD8,  z2_dir_pin=PC6,   z2_enable_pin=PC7,   z2_uart_pin=PD10,   z2_diag_pin=null,
  # Voron 2.4 support
  #z3_step_pin=PC9,	 z3_dir_pin=PC8,	 z3_enable_pin=PD1,	  z3_uart_pin=PD0,   z3_diag_pin=null, 
  e_step_pin=PD7,    e_dir_pin=PD6,    e_enable_pin=PF10,   e_uart_pin=PF9,    e_diag_pin=null,   e_heater_pin=PE3,  e_sensor_pin=PA1,
  stepper_spi_mosi_pin=PA7,  stepper_spi_miso_pin=PA6,  stepper_spi_sclk_pin=PA5,

Z2 is like you said above but when i do a stepper_buzz i get this error
"Unable to read tmc uart 'stepper_z2' register IFCNT"

looxonline, 2023-08-22:
@NAPCAL @azhago @sergelhr @RezezRX79 @12voltnick @Synapsis181 

This issue was caused by the input voltage tolerance of one component connected to the 6th stepper driver being right on the edge of the 3V3 level that the MCU outputs. Some components accepted the level from the MCU and others did not.

If you guys have been affected by this issue then BTT would like to offer you a 15% compensation off the cost of your board. Please contact me using the "bttuniversity" handle on discord.