bigtreetech/Manta-M8P #48: Neopixel do not work

MikeEitel, 2023-03-13:
I can't get any neopixel running on my mp8/cm4. Using klipper.  
Rest works like a charm. 

I defined both channels in printer.cfg with the probably correct pins. See them in the dashboard. There I can change colors. Also using macros change the color in dashboard. 
But straps stay dark. 
Measuring (both) dedicated pins with an oscilloscope shows no signal. Always stays ground.
Testing with a resistor to 5v doesn't show signal neither..

I'm wondering if there are wring pins in doc or I need special firmware.
Any ideas?

Thx in advance

MikeEitel, 2023-03-15:
Any idea?
Thx

qiyuanlu, 2023-04-29:
I have the same issue are you using an ebb2209 by any chance?

MikeEitel, 2023-04-29:
No
But I gave up to use neopixel.

As support from bigtreetech sucks

nineangel, 2023-06-06:
Double-check your word version. they changed PIN ID in v1.1
also make sure if led strip has lots of LEDs you need additional power connected in the middle or turn the brightest down. otherwise it draws too much current and shuts down the board.

ecob81, 2024-05-19:
Has anyone got this working yet? I can't for the life of me get neopixels working off the m8p. The toolhead neopixels work ok that are running off a Canbus board however there's nothing but darkness when trying to use the headers on the main board. Seems like the neopixel signal isn't even getting generated.

sajrashid, 2024-09-10:
I can't get Neopixels to work from RGB port (Voron [Daylight on a stick](https://www.aliexpress.com/item/1005005913975209.html?spm=a2g0o.order_list.order_list_main.65.794718023jJLQv))
first LED flashes for 0.2 seconds on boot or while changing colours from Klipper, I'm sure it's taunting me  
Other than that no cigar 🚭 I'm afraid 😨.

Voltage is **OK**
GND is **OK**
Data - **no scope :(**

```
[neopixel my_RGB]
pin:PD15
chain count:18
color order: GRB
initial_RED: 0.2
initial_GREEN: 0.2
initial_BLUE: 0.2
```

@bigtreetech  help please ?

looxonline, 2024-09-10:
> I can't get Neopixels to work from RGB port (Voron [Daylight on a stick](https://www.aliexpress.com/item/1005005913975209.html?spm=a2g0o.order_list.order_list_main.65.794718023jJLQv)) first LED [...]

Some have reported that a series resistor helped them. Could be related to a long wiring path and that path needing a bit of damping due to parasitic inductance.

sajrashid, 2024-09-10:
> Some have reported that a series resistor helped them. Could be related to a long wiring path and that path needing a bit of damping due to parasitic inductance.

Hi, Just tried 220K on the data line, exactly same behaviour I can make it flash for s fraction of sec while changing colour using the wheel in mainsail.

Seeing if I can find 100.

As for the wiring thing is I've tested from 1.5m of wire to less than 0.5m and nothing.

**Thankyou for the suggestion**

Contemplating, might just use an esp32 with moonraker 🤔

looxonline, 2024-09-11:
> Hi, Just tried 220K on the data line, exactly same behaviour I can make it flash for s fraction of sec while changing colour using the wheel in mainsail.
>
> Seeing if I can find 100.
>
> As for the [...]

220k is way too high. You want something between 100 and 330ohms.

sajrashid, 2024-09-17:
> 220k is way too high. You want something between 100 and 330ohms.

Sorry, My bad I meant to write 220R - I also found a 100 ohm resistor no joy

This is what I have tried board is Manta M8p v2 

- Inline resistor (data line) 220 and 100 ohms  😔 
- Capacitor small value on the 5v line  😔 
- 5v 6a SMPS with a shared ground 😔 

I got to light up accidently as the ground was loose, but only about half lit up and did not respond to software changes (apart from a slight flicker)

So has the v2 gone backwards in the LED game

This whole thing  is suss - I mean if you sold an LED controller that did not work in how many situations ?