bigtreetech/Manta-M8P #52: No power to CB1 VIA M8P Led on CB1. CB1 Powered and boots Via USB.

Zaskeller, 2023-03-26:
Having an issue where the CB1 will not boot via power from the M8p, M8P shows 12,5,3.3 volts are powered via the indication Led.

The CB1 does boot via USB and activates wifi and UI. but doesn't boot without USB power pass through. LED2 is powered on CB1 in both modes. there is also no LED for PIEFAN.

Zaskeller, 2023-03-26:
After Testing found that if the USB pass through for power is bridge then you turn on the M8P power, it will power the PIFAN pin once unplugged from the USB. this seems to jump start the board where power from just the m8p can't.

Zaskeller, 2023-03-26:
When Powering on the m8p the PIFAN pin may flash, and the pi will only boot when this led seems to be on.

dawesy72, 2023-12-20:
I have the same situation although I haven't tried booting from usb only yet. 
Are you putting dip switch 3 or 4 to on or both. I've tried both as card slots too.