bigtreetech/Manta-M8P #57: [Question] Manta M8p , CB1 and powering CB1.

cruzer619, 2023-04-06:
Is it possible to powerup the CB1 without power up the mainboard.  I'm used to having Pi on all the time.  So I can remote power up the printer via telegram or other remote tools and print remotely.  Is this possible.  I am aware of VUSB jumper but I have a feeling still powers the mainboard.

Roguyt, 2023-04-07:
I'm also interested in this option. It's a bit annoying to have to power off the PI and then power off the printer.

its-romo, 2023-05-26:
No there is no option to power the CB1 board separately from the mainboard, they are designed to be treated as a single component. 

If you want to use the CB1 as a standalone device, you will need to purchase a PI4 Adapter from BTT or another vendor.