bigtreetech/Manta-M8P #58: Loss of power on m8p

Lazybones2319, 2023-04-26:
G’day all,
I recently got a mantra m8p with cb1 for my e5+ upgrade (overkill to the max) I managed to compile all the Pi os image and firmware for the boards got everything hooked up and began setting the config for my printer. Once I was done for the day I turned in went to bed with a smile on my face because everything was starting to work as it should. Well that smile turned to anger the next day when I switched the power back on for the printer and waited, waited a bit more then about 30 minutes later (after making a coffee and toast) I noticed the power LED near the hdmi plugs was non existent… in my mind I thought oh it must not have booted properly, so again off goes the switch and let it sit while I had a couple games of xdefiant and apex legends. Well switched it back on and waited 10 minutes…. Still…. No power to the mcu but everything else had power so I thought oh well I’ll jump onto my browser to open mainsail… nope nothing so I powered down the printer disconnect all the attached cables turned it back on and waited again with no luck…. 

rage boiled inside me as I’ve put so much effort into reviving this printer and making it how I want it. No harm came to any devices or printers unless you call running my ender 3v2 for a 29hr print harmful…. 

I have made multiple attempts to contact big tree tech now to try and find a solution or resolution of some sort… I may just be impatient as they haven’t responded yet. 

my question is to the community, how many have experienced this or something similar? And what was the outcome as I don’t plan on spending money on another board (although it seems that may be the fastest way to get my printer operational again)

Rlinhos, 2024-02-26:
Did you find the solution? I'm having the same problem and the fuses are working perfectly. I'm desperate because I don't know what else I can do and the board is new.

dasboot58, 2024-08-08:
I have a similar issue... Yesterday, before I connected the extension cables to the rest of the machine, I was able to power up without issue. I am now getting only the 12V led. The board power led comes on briefly when first powered. If anyone has any advice, I would appreciate it. I am used to glass fuses, so if there is a particular board fuse I should check, I would appreciate the advice.

bigtreetech, 2024-08-08:
This can happen if both USB power and the main 24V power are applied to the board at the same time and the USB jumper is inserted. It would result in the two 5V supplies fighting it out and it could cause damage.

There are also other possible sources where damage could be caused but in general it would be from overloading the 5V regulator on the M8P. Other than the USB issue mentioned, it may also come from a wiring issue causing a short to GND or perhaps feeding another 5V source back into the M8P 5V rail.

dasboot58, 2025-02-07:
I was struggling with this for a while. I finally realized it was my end stops. The wiring my Ender 5+ (going from a SKR mini e3) required moving the contact in the “+5V” position to the “PF_” position.