bigtreetech/Manta-M8P #59: CSI port on underside of m8p

philgreenh22, 2023-05-02:
hi all

i have a manta M8P with CB1. i am trying to add a camera to it but as soon as the camera is inserted the board will not post.

i have searched for the /dev/video* but have no video folders to list.
i have searched for /dev/usb/by-id and only find the klipper mcu
i have searched  for /dev/serial/by-path but never find nything but the klipper mcu.
i have enabled legacy video in the bios

what have i missed???

please before i become completely bald....

any help appriciated

Phil
[klippy (3).log](https://github.com/bigtreetech/Manta-M8P/files/11376637/klippy.3.log)

looxonline, 2023-05-03:
The CB1 does not support CSI or DSI. Only USB cameras and HDMI/SPI displays are supported.

philgreenh22, 2023-05-04:
lol

shame i didnt see that before opting for a pnp solution.

at least i can chill now i know its not a me issue.

what is the point of including these ports i wonder?

thanx for the info. :)

closed 2023-05-04 by philgreenh22