bigtreetech/Manta-M8P #61: i2c

albielew, 2023-05-16:
Which pins support i2c?
I want to use a BDsensor that communicates using i2c and it's not in the manual

looxonline, 2023-06-05:
There is no i2c bus on the motherboard.

albielew, 2023-06-05:
There actually is, I've used it.
Took me ages to find it but it's there

looxonline, 2023-06-05:
> There actually is, I've used it. Took me ages to find it but it's there
> […](#)

There are a few options that could be used but none of them are intended for I2C. You would be hijacking another function.

real-artswan, 2025-03-07:
So, in the year 2025, on the Manta8P v2, how to use i2c? For example these pins from the official PDF? 

[image]

What I've tried: 
Looked in the klippy.log where I found

```
Loaded MCU 'mcu' 123 commands (v0.12.0-451-g730e5951b / gcc: (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027] binutils: (2.35.2-2+14+b2) 2.35.2)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c1_PB6_PB7=PB6,PB7 BUS_PINS_i2c1_PB8_PB9=PB8,PB9 BUS_PINS_i2c2_PB10_PB11=PB10,PB11 BUS_PINS_i2c3_PA8_PC9=PA8,PC9 BUS_PINS_spi1=PA6,PA7,PA5 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_spi3a=PC11,PC12,PC10 BUS_PINS_spi4=PE13,PE14,PE12 BUS_PINS_spi5=PF8,PF9,PF7 BUS_PINS_spi5a=PH7,PF11,PH6 BUS_PINS_spi6=PG12,PG14,PG13 CANBUS_BRIDGE=1 CLOCK_FREQ=400000000 MCU=stm32h723xx PWM_MAX=255 RECEIVE_WINDOW=192 RESERVE_PINS_CAN=PD0,PD1 RESERVE_PINS_USB=PA11,PA12 RESERVE_PINS_crystal=PH0,PH1 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
```

PA8 and PC9 looked like what I need so I tried config
```
[temperature_sensor chamber]
sensor_type: AHT10
#   Also use AHT10 for AHT20 and AHT21 sensors.
#i2c_address:
#   Default is 56 (0x38). Some AHT10 sensors give the option to use
#   57 (0x39) by moving a resistor.
i2c_mcu: mcu
i2c_bus: i2c3
#i2c_speed:
#   See the "common I2C settings" section for a description of the
#   above parameters.
#aht10_report_time:
#   Interval in seconds between readings. Default is 30, minimum is 5
```

i2c3 was not found by Klipper after reboot then I enabled it in the overlays via armbian-config and then system halted on boot almost all the time (~ 19 times from 20). So that I had to switch power until I got OS booted up and disabled that overlay.

real-artswan, 2025-03-07:
I also tried software i2c

```
[temperature_sensor chamber]
sensor_type: AHT10
#   Also use AHT10 for AHT20 and AHT21 sensors.
#i2c_address:
#   Default is 56 (0x38). Some AHT10 sensors give the option to use
#   57 (0x39) by moving a resistor.
i2c_mcu: mcu
# i2c_bus: i2c3_PA8_PC9
i2c_software_scl_pin: PA8
i2c_software_sda_pin: PC9
#i2c_speed:
#   See the "common I2C settings" section for a description of the
#   above parameters.
#aht10_report_time:
#   Interval in seconds between readings. Default is 30, minimum is 5
```

It seems no problem with booting but getting `MCU 'mcu' shutdown: I2C START NACK`

real-artswan, 2025-03-07:
Hm, it seems it was some connection issue, now it works with software i2c and you don't need any additional overlays.

oskarekxy, 2025-04-05:
@albielew Which pins did you use exactly? I have a Manta M8P V1.1. Did you take the pins from EXP1/2 or the display connector