bigtreetech/Manta-M8P #62: M8P - SPI mode - TMC5160 (Mellow external)

surikatec, 2023-05-19:
Hello guys,

why for me didn't work HW SPI mode pro TMC5160 (Mellow external)? Jumpers on motherboard have correctly. Works only SW SPI.

I had a version M8P 1.0, changed for 1.1 and still didn't work as I wish. :(
[image]

Thanks.

Michil783, 2023-06-08:
Have the same issue with TMC2130. SW SPI is working but nor with spi-bus parameter in config set to spi1

BR Michael

staticanime, 2023-09-08:
Try the below, that got 5160's working for me over SPI:
```
[tmc5160 stepper_x]
cs_pin: PF13
# spi_bus: spi1
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
```