bigtreetech/Manta-M8P #64: USB C not working

rrgmaly, 2023-05-25:
When I hook up the HDMI7 to either of the USB 2.0 ports the HDMI unit works fine, If I connect it to the USB C port nothing happens (the jumper is off, btw), the USB C port also does not seem to have any power going out.  Is there a setting for the USB C port that I am missing or is the USB C port only for power in?

its-romo, 2023-05-26:
Try adding this line to the `/boot/config.txt` file

`dtoverlay=dwc2,dr_mode=host`

Let me know if it helps :)

rrgmaly, 2023-05-26:
Already added it after flashing the board.  The two USB 2.0 are working, it is only the USB C that is not working.

rrgmaly, 2023-06-07:
After some research, it appears that the USB C port is only for power and EEPROM flashing.

freakydude, 2025-03-04:
Tried it today on M8P V2.0 to connect my webcam - but as you said - it's not working for me either 👎🏼

looxonline, 2025-03-04:
The USB-C port is only for flashing the CB/CM and for serial debug. This information is available in the manual. USB-A ports are for peripherals.

closed 2025-03-04 by looxonline