bigtreetech/Manta-M8P #67: OS Intermittently Freezing and Corrupting SD Card

420noscope-exe, 2023-06-09:
I have a manta M8P mainboard with a CB1 installed onto it, running the provided klipper image.  I always leave my machine idling, powered on at all times.  Randomly, after about a week of idling, the web interface crashes.  I am unable to access it at all. 
 I cannot ssh into the device either.  The only solution is to unplug the printer and plug it back it.  This has a high probability of corrupting my SD card, and I've had to re-image it about 4 times now.  Where do I start with this?

closed 2023-06-13 by 420noscope-exe

Peoplles, 2025-09-26:
Hello, I am having the same issue except it's daily.  What solution did you arrive at?

420noscope-exe, 2025-11-14:
@Peoplles https://github.com/bigtreetech/CB1/issues/125