bigtreetech/Manta-M8P #68: Dead USB Hub

timmit99, 2023-07-03:
# USB Hub dead

## Hardware:
- BTT M8P
- Raspberry Pi CM4 2GB ram, 32GB eMMC
- Voron 2.4 printer

## Issue:
* M8P's onboard USB hub has died 48 hours after unboxing.

# Diagnosis:
I was going through my initial setup, had gotten the boards serial number for klipper, as well as another board via `lsusb` and `ls /dev/serial/by-id/*` and all was working fine. All of the sudden, after a `FIRMWARE_RESTART` to update a config value, the MCU couldn't connect.

Sure enough, upon further evaluation, LSUSB returned only a single item, the Linux root hub. It no longer saw the onboard Hub IC or thee camera/controllers connected to the board:
[image]

To check if the CM4 was damaged or the M8P, i powered down and swapped the CM4 over to my BTT PAD5 and sure enough, there, it detected all the devices plugged in:
[image]

This shows that the CM4 is working as intended and is not the issue. Inspection of the board shows no signs of damaged chips, burnt parts, lifted traces, etc.

looxonline, 2023-07-03:
First time I am hearing of this. I will skim through the schematic to see if there is anything that may have caused it.

timmit99, 2023-07-03:
Thanks, let me know what you want me to do for testing. Happy to try and diagnose this issue. I've done visual inspection with no obvious signs of damage and can upload more photos tonight.

timmit99, 2023-07-03:
## Board Overview Top
[image]
## CM4 Area Top
[image]
## CM4 Overview Bottom
[image]
## USB Hub IC
[image]

The USB Hub IC looks suspicious.

Twdiap, 2023-11-08:
I have the same problem, BTT told me the board is dead and refuse warranty...

looxonline, 2024-07-04:
> I have the same problem, BTT told me the board is dead and refuse warranty...

Hey there! I have just been speaking to timmit about this and I thought that I would come and search to see if anyone else was having this issue and I found you. I would like to offer to replace your M8P if you are still in possession of it and can show me the lsusb results. Please send me a mail at luke@bigtree-tech.com if you would like to take me up on the offer.

pokeygithub, 2024-07-28:
My problem is slightly different.  Mine is missing the 5vDC  On the usb ports.