bigtreetech/Manta-M8P #70: Network issues. Cb1 v2.2. Ratos

marcged, 2023-07-08:
Hi there. I've had no trouble up until I had to turn the power off to my garage. Now when I turn my printer in, it connects to the network but it doesn't allow me to do anything. I can't access the web interface, I can't Ssh into it either. Ethernet or usb wifi dongles don't make a difference either. Is this a cb1 issue or a ratos issue?

Thanks,
Marc.

pOmelchenko, 2023-07-19:
>  Is this a cb1 issue or a ratos issue?

Did you try to connect directly to the motherboard? Do you have CB1 PI4B Adapter to check problems with M8P?

marcged, 2023-07-19:
> 

So, It turns out my board has a loose sd card holder. It must have got corrupt because I cannot view any files (printer.cfg etc) in the sd card when  I try and read from it. I can access the system portion but not the part that stores cfg files. I just reflashed the board. I lost 3k hours of printing history but all I care about is I didn't need to buy another board.

pOmelchenko, 2023-07-21:
I use cb1 with flashed OS onto emmc so I forgot about sd cards, sorry