bigtreetech/Manta-M8P #71: DSI and CSI not working at the same time

Peviox, 2023-07-08:
im using my M8P with a CM4 4gb ram and 8gb eMMC with a PITFT 2.0 with DSI and a cheap raspicam over CSI. 
If i connect the display alone it works fine and displays stuff. If i connect the camera as well, the camera works but the display stops displaying or reading touchinput, not even teh backlight turns on. 
i installed the DSI driver according to the instruction. 

waht could be the problem and solution here?

Pippowicz, 2023-08-26:
The same for me, 

CM4, 8GB, 32Gb EMMC, no WiFi

hadesz, 2023-10-08:
I have the same problem. I can confirm, there's an i2c error in system log...

Tuxedo-VE, 2025-01-20:
I had the same problem. This is caused by an error in the schematic: the pin assignments for the 15-pin DSI header on the Manta are inverted when compared to the normal pin assignments on a Raspberry Pi's CSI header. Because of this, you cannot use the 15 conductor Type 2 FFC flat cable that comes with the Pi CAM module (exposed contacts are on opposite sides of the flat cable). You must use a Type 1 FFC cable with exposed contacts on the same side of the cable.

Tuxedo-VE, 2025-01-24:
> Yes, you are right, found the solution too b ut forgot to post.
> 
> Type 1 FFC Cables are difficult to get, so have fun finding one :(

Ever heard of Amazon? I'm looking at one right now...
https://www.amazon.com/uxcell-Flexible-Cable-Ribbon-Player/dp/B07RWRX6YT/ref=sr_1_1?crid=D2JAX5VR78BQ&dib=eyJ2IjoiMSJ9.5hjc0EbpuB94tDyZLcCAIvswJTHwSOZhpuT-pKQwEHO11kxyPPO2eQu5iDw9Z2QL6LgxN30rVrw4W19BdS2o9k9edX_-rzBKuDtDVaZN7u6V39uMHwsl8Tnv7foHkd9qPSgpYV8l8YxF2ovhBfOxoIKYOWuFwNYH-TGbKloOuDG53ZKoNvE3aKqBnlAeU7BIpKW8HBH9XN7v2oDnXSjKANvm4rWucuuNmmn2Q0DtfSE.5dHASmHJG8Mh6grgaXsaNJ6T1l_Ss5f7pXJ_JechCLY&dib_tag=se&keywords=FFC+type+a&qid=1737728957&sprefix=ffc+type+a%2Caps%2C83&sr=8-1

hazama6, 2025-04-11:
You may not need a new cable. The m8p V2.0 has a connector that works both ways. Try flipping it and do a continuity tests with a multimeter

Pippowicz, 2025-04-11:
@Tuxedo-VE: You DO realize my Post is from one year ago AND i am from Germany so no luck here.

Either way it is a poor Design Decision from Bigtreetech.

Tuxedo-VE, 2025-04-11:
> You may not need a new cable. The m8p V2.0 has a connector that works both ways. Try flipping it and do a continuity tests with a multimeter

Thanks for the tip. I wish I had known that while I was building my printer. BTT should have mentioned this in their manual. That was a while ago... I bought and installed a reversed flat cable and everything works fine. It's no big deal though, the cable wasn't expensive. I'm just happy I didn't fry the CSI port on my Manta while hooking it up with reversed connections.

bigtreetech, 2025-04-11:
Sorry about this guys. We will update the manual to make mention of the reversible connector.

hazama6, 2025-04-18:
> Thanks for the tip. I wish I had known that while I was building my printer. BTT should have mentioned this in their manual. That was a while ago... I bought and installed a reversed flat cable and [...]

Had the same concern. Thought i had fried the port. Managed to ge things working eventually though