bigtreetech/Manta-M8P #74: Sensitivity of 2209 sensorless homing

deonkretch, 2023-07-21:
Configuring sensorless homing on a trident with M8P + CM1 + SB2209.

During the setup of the sensorless homing I go through the recommended process to start the test by setting the sensitivity to max (255). In the manuals the toolhead should stop immediately, but in my case it keeps moving and slams into the side of the gantry.

This is the command:

`SET_TMC_FIELD FIELD=SGTHRS STEPPER=stepper_x VALUE=255`
followed by:
`G28 X0`

Printer config:

`#####################################################################
#   X/Y Stepper Settings
#####################################################################

##  B Stepper - Left
##  Connected to M1

[stepper_x]
step_pin: PE2
dir_pin: !PB4
enable_pin: !PC11
microsteps: 16
rotation_distance: 40
# endstop_pin: ^PA3
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: 0
position_max: 235
position_min: 0
homing_speed: 40 #Max 100

homing_retract_dist: 0
homing_positive_dir: false

[tmc2209 stepper_x] #M1
uart_pin: PC10
diag_pin: ^!PA3
driver_SGTHRS: 255
run_current: 0.650
stealthchop_threshold: 999999`

Pins are bridged on the board as indicated in the BTT Manual to enable sensorless homing.

Any suggestions?

heemanshoe, 2023-09-11:
Did you figure this out? I am running into the same issue with tmc5160 (I am using SPI instead of UART though)

deonkretch, 2023-09-12:
I decided to use normal end-stop switches for this build.

suprchunk, 2024-06-10:
"driver_SGTHRS: 255"

Try a LOT LOWER, like 127, and adjust from there. 255 is the supposed most sensitive.