bigtreetech/Manta-M8P #78: CB1: '/dev/serial/by-path/': No such file or directory

Morogorn, 2023-09-13:
**I lost serial connection.**

root@BTT-CB1:/home/biqu/klipper# ls /dev/serial/by-path/
ls: cannot access '/dev/serial/by-path/': No such file or directory

---
wget https://github.com/bigtreetech/Manta-M8P/blob/master/V1.0_V1.1/Firmware/M8P_bootloader.bin

dfu-util -d ,0483:df11 -R -a 0 -s 0x8000000:leave -D ./M8P_bootloader.bin

**Does not work**
---
**After click button [boot] + [reboot]:**
root@BTT-CB1:/home/biqu/klipper# dfu-util -d ,0483:df11 -R -a 0 -s 0x8000000:leave -D ./M8P_bootloader.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08000000, size = 5045
Download        [=========================] 100%         5045 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status
root@BTT-CB1:/home/biqu/klipper#
---
**Device "0483:df11" apears.**

root@BTT-CB1:/tmp# lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
_Bus 001 Device 003: ID 0483:df11 STMicroelectronics STM Device in DFU Mode_
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
-----

But, if reset:
"0483:df11 STMicroelectronics STM Device in DFU Mode" disapear.

@bigtreetech , how solve this?

Fionoz, 2024-03-18:
Any solution?

charchess, 2024-03-18:
Hi, 

I had the exact same issue, they made me flash with the ST link and it went away

Regards

Morogorn, 2024-04-22:
> Hi,
> 
> I had the exact same issue, they made me flash with the ST link and it solved the issue

Can share the link or procedure?

charchess, 2024-04-22:
[M8PV2 How to upate bootloader with ST LINK.docx](https://github.com/bigtreetech/Manta-M8P/files/15063599/M8PV2.How.to.upate.bootloader.with.ST.LINK.docx)

Here is the procedure they send me, hope it'll help

barrymathes, 2024-07-12:
@charchess   I looked at that doc but it doesn't show what the matching pin connections are between the board and the ST-Link?  i list the ones on the ST-Link but the wire colors don't match in the other pic and Biqu doesn't list the pins in the pinout description on any of their documents.