bigtreetech/Manta-M8P #85: M8P v2.0 boot loader issue and problems configuring CAN

Ram-85, 2023-09-23:
I am having issues configuring CAN BUS with this board, it seems the bootloader is trashed and I unable to flash firmware via sd card

abo974, 2023-10-21:
I have same issue. There's different information from EEB Can and M8P config file document ( we have to use Canbus PB0/PB1 or PD0/PD1 ?). I follow the EBB SB2240 guide and now I can't upload any firmware on M8P... and it doesn't work :/

BobArctor82, 2023-11-01:
someone have a solution ? same probleme after the ebb sb2240 guide

abo974, 2023-11-02:
> someone have a solution ? same probleme after the ebb sb2240 guide

You can flash the board without the sd card... Mine work great, except the undervoltage of the TMC2240 :/.
On both the SB2240 and M8P you've to flash Canboot and Klipper, copy the uuid in your printer.cfg. When you flash the board you can't access to the serial port... 
I read a lot of tutorial, and you can find some command in the file below. Make sure you have select communication port PD0/PD1  to flash the Manta board and select communication port PB0/PB1 for the SB2240.
[flashing_Manta_v2_SB2240.txt](https://github.com/bigtreetech/Manta-M8P/files/13236982/flashing_Manta_v2_SB2240.txt)

nero-dv, 2024-01-14:
I had the same problem. I ended up messing up the bootloader on the M8P, and my CM4 could not see the M8P board when I ran the lsusb command. What I ended up doing was flashing the bootloader available in the repo ([link](https://github.com/bigtreetech/Manta-M8P/blob/master/V2.0/Firmware/M8P_V2_H723_bootloader.bin)) with dfu-util

1. Press and hold the boot button of M8P and then click the reset button to make M8P boot into DFU mode.
2. While logged into the CM4/CB1, run `lsusb` to check the MCU's USB-DFU address.
3. Run the following command on the Terminal of CB1/CM4 to write the bootloader to the chip:

`sudo dfu-util -d ,0483:df11 -a 0 --dfuse-address 0x8000000:force:mass-erase -D ./M8P_V2_H723_bootloader.bin`

After this, I set up the can0 interface:

`sudo nano /etc/network/interfaces.d/can0`

Add the following:

```
allow-hotplug can0
iface can0 can static
    bitrate 1000000
    up ifconfig $IFACE txqueuelen 1024
```

Then I power cycled the device and then logged back in via SSH, then I ran:

```
cd ~/klipper
make menuconfig
```

I used the following settings:

> [*] Enable extra low-level configuration options
>     Micro-controller Architecture (STMicroelectronics STM32)  --->
>     Processor model (STM32H723)  --->
>     Bootloader offset (128KiB bootloader (SKR SE BX v2.0))  --->
>     Clock Reference (25 MHz crystal)  --->
>     Communication interface (USB to CAN bus bridge (USB on PA11/PA12))  --->
>     CAN bus interface (CAN bus (on PD0/PD1))  --->
>     USB ids  --->
> (1000000) CAN bus speed
> ()  GPIO pins to set at micro-controller startup (NEW)

Then I ran make, rebooted into DFU mode again, then:

`make flash FLASH_DEVICE=0483:df11`

After that finished, I reset the MCU by clicking the reset button on the board and verified that the M8P showed a CAN address:

`python3 ~/klipper/lib/canboot/flash_can.py -q`

After this, I followed a [guide](https://mpx.wiki/flash-m8p-v2-and-ebb-sb-toolboard) on flashing Katapult and Klipper onto the tool-head board I'm using (SB2209 RP2040). 

One thing to note is that a lot of guides mention shorting the 120R terminator resistor, but they don't specify that it has to be done on **_both_** the Manta and the tool-head PCB.

Drath1995, 2024-04-08:
> I had the same problem. I ended up messing up the bootloader on the M8P, and my CM4 could not see the M8P board when I ran the lsusb command. What I ended up doing was flashing the bootloader [...]

Twice now you've saved my bacon with these instructions. I cannot thank you enough, praying theres not a 3rd time, but if there is i've at least download this page and got it backed up in multiple locations

charchess, 2024-04-08:
Hi, 

you need 2 resistor terminators, no less, no more. One should be on the motherboard and the other on the other side (EBB)

I had issues with that CAN thing and followed https://canbus.esoterical.online which is really comprehensive and got juste the pictures you need. It then worked like a charm, it's really the only guide you need for the CANbus, 

hope it'll help

Space-Stick, 2025-03-06:
Thank you Nero-DV, was looking to unbrick my M8P and you helped me not only do that but get can working at the same time. Legendary.