bigtreetech/Manta-M8P #86: M8P V1.1 - Not going into DFU mode.

deonkretch, 2023-10-01:
Manta M8P refuses to go into DFU mode.

The board is powered via USB with the CB connected. I can SSH to the CB1.

**I have tried:**

- Hold BOOT, Click RESET (the run 'lsusb'), nothing.
- Removed CB1, downloaded M8__bootloader.bin. Renamed to firmware.bin and loaded to the MCU SD card slot. Boot the M8P and left for a couple of minutes. The firmware.bin has not been renamed to firmware.cur
- Remove power. Hold BOOT, connect USB power, release boot. Nothing
- Removed the CB1 and reseated it to ensure that it is connected.

When SSH to the board and running the `lsusb` command this is the output:

`biqu@BTT-CB1:~$ lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

**Question:**

1. Is there a way to force reset the M8P?
2. Is there another way to get the M8P into DFU mode?

SarcasmRules, 2023-10-22:
Same problem here, need to get it into dfu mode to reset/reflash the mcu/booloader. 
Anyone???

celticapatchanka, 2023-10-22:
Having the same issue.
The CB1 stopped communicating with the mcu and no devices appear anymore when I run the command ls /dev/serial/by-id/ so I wanted to try and reset the mcu

deonkretch, 2023-10-23:
> Same problem here, need to get it into dfu mode to reset/reflash the mcu/booloader. Anyone???

Nothing yet...

ijanda, 2023-11-20:
I have the same problem. I've been trying to install the Manta m8p V2 for three days. No chance - it won't go into DFU mode. It is therefore not possible to flash the Klippy.bin because it also does not work via SD card.

elindgren, 2024-02-04:
Same problem here! Have tried flashing via the SD-card method but no luck (pressing the reset button does nothing), and nothing shows up under `/dev/serial`.

SarcasmRules, 2024-02-05:
I don't think that BTT cares enough to fix this issue, they rather have you buy a new one? 🤷🏼‍♂️
Right now it's a paperweight, not a cheap one at that

elindgren, 2024-02-06:
> Same problem here! Have tried flashing via the SD-card method but no luck (pressing the reset button does nothing), and nothing shows up under `/dev/serial"

"Solved" the issue by rolling back to a previous version of klipper, using the rollback functionality of `kiuah`

Fionoz, 2024-03-18:
@deonkretch Did you manage to fix it?

SarcasmRules, 2024-03-19:
> > Same problem here! Have tried flashing via the SD-card method but no luck (pressing the reset button does nothing), and nothing shows up under `/dev/serial"
> 
> "Solved" the issue by rolling back to a previous version of klipper, using the rollback functionality of `kiuah`

What version did you go back to?

elindgren, 2024-03-19:
> > > Same problem here! Have tried flashing via the SD-card method but no luck (pressing the reset button does nothing), and nothing shows up under `/dev/serial"
> > 
> > 
> > "Solved" the issue by rolling back to a previous version of klipper, using the rollback functionality of `kiuah`
> 
> @elindgren What version did you go back to?

I don't remember the specific version from the top of my mind, but I went back to the second-to-last release tag available in the git repository. You can find the tags by running `git tag -l`. I then took the git hash of the release and rolled back to that commit

kmjr20, 2024-07-21:
I am having the same problems with all of a sudden the machine can't see the mcu. attempts to reinstall everything have failed, and you now cant even get it to go into DFU mode. I am frustrated as it started with not being able to set up the 2 z motors getting a UART pin error no matter what I did. Then after rebooting when I added the new TFT70 V3, the mcu stopped working and running any command about serial port results in not finding any, nor the MCU. The manual makes a lot of assumptions that you are a programmer and can fill in the blanks in the directions. Also, the TFT70 v3 shows EXP cables, but there are none on the M8P V2 board. Nothing about how to get it to run. Suggestions about rolling back to previous versions to fix the problems are not realistic as the V2 comes with V2 software. No previous versions for the V2 are on the github website.Does anyone know what I can do. I have been trying to upgrade my TRONXY VEHO 1000 printer, which never worked right out of the box. I have tried reaching out to BTT, and as of yet I have never been able to get anyone or responses from the emails.