bigtreetech/Manta-M8P #87: Suggested document update for rpi Boot mode

justinhardin, 2023-10-05:
In the file: BIGTREETECH MANTA M8P V1.0&V1.1 User Manual.pdf
On page 28
Step 2
It says 'Push the DIP switch 4 (USB OTG) and 3 (RPIBOOT) to ON to enter BOOT mode.'

I discovered that there is another step possibly required...
there is a micro-switch to the right of the CM4 slot, between the USB C connector and the HDMI 1 connector, that must be in the correct position. (In the USB-OTG position I believe) before the Raspberry CM4 eMMC will show up in RPI Boot.

I spent several hours troubleshooting this, thought I had a bad CM4 board, did a bunch of debugging...then flipped that switch and rpi boot brought the eMMC storage right up.

igiannakas, 2024-03-23:
THANK YOU!!🙏🏻 

been hitting my head against a brick wall with this. Flipped that switch to the uart to USB setting and it was recognised straight away.

vikcious, 2025-06-02:
I've been to hell and back till the moment I discovered on my own that damn switch and (out of desperation!) I've switched its position to **"UART-to-USB"** from the default "USB-OTG" and MIRACLE: CM4 eMMC was recognized by rpiboot.exe and I was able to flash the Mainsail 2.0 x64 image.

bigtreetech, 2025-06-02:
It's the opposite. It should be in USB-OTG mode but maybe the drawing makes the dark and light spots seem like it is in the opposite orientation.

vikcious, 2025-06-02:
> It's the opposite. It should be in USB-OTG mode but maybe the drawing makes the dark and light spots seem like it is in the opposite orientation.
Could you please confirm which position this really is. It might be a great deal of help for the others.
So for me, the switch was on the opposite position, so it came by default set. And now, only in this position my CM4 eMMC eventually became available to be flashed. Thank you!
[image]

bigtreetech, 2025-06-02:
No, I think it is the light and dark spots in the manual which are confusing. The dark indicates the switch position so right is UART to USB and left is USB-OTG.

vikcious, 2025-06-02:
Ok. I would read it the opposite way... since white is the smaller surface on the magenta area BUT thanks for the clarification. 
I've been banging my head against the wall for the past 48 hours not knowing what i do wrong and planned to even return my Manta M8P to Amazon on such grounds.
Can I kindly ask you to **add a phrase detailing this pre-requisite of this switch being in the right position** for the CM4 eMMC flashing (page 24). There are, as of today, zero references inn the whole manual about this switch and trust me other forums are filled with people fumbling on not being able to get their CM4 eMMC recognized and flashed. **Thanks a lot!**

bigtreetech, 2025-06-02:
We will need to update the manual with a drawing that is less confusing. I will get the guys who work on the manuals to do this as soon as they are back from the dragon boat festival tomorrow.

vikcious, 2025-06-02:
> We will need to update the manual with a drawing that is less confusing. I will get the guys who work on the manuals to do this as soon as they are back from the dragon boat festival tomorrow.

**Thanks!** So the update needs to target:
1) the drawing explanation & colors need be updated (drawing on page 8)
2) this phrase, on page 24, needs to be updated from:
_"2. Toggle switches 4 (USBOTG) and 3 (RPIBOOT) to ON to enter BOOT mode."_ should read **_"2. Toggle switches 4 (USBOTG) and 3 (RPIBOOT) to ON to enter BOOT mode AND confirm that SW2 switch is in "USB-OTG" position and not in "UART-to-USB"."_**