bigtreetech/Manta-M8P #92: Homing failed due to printer shutdown

fish1942, 2023-11-06:
I'm getting "Homing failed due to printer shutdown."  or "Unable to write tmc spi 'stepper_x' register GCONF" when trying to home custom Vorom 2.4. I've checked motor wiring and all checks out 
Running TMC5160 drivers
Hermit Crab/canbus 

I'm stuck 

printer.cfg 

# This file contains common pin mappings for the BIGTREETECH Manta M8P
# To use this config, the firmware should be compiled for the
# STM32G0B1 with a "8KiB bootloader" and USB communication.

# See docs/Config_Reference.md for a description of parameters.
[include Hermit Crab.cfg]
[include nozzle_scrub.cfg]
[include mainsail.cfg]
[virtual_sdcard]
path: /home/biqu/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

[mcu]
canbus_uuid=63514f1fd4c8
##	[X in MOTOR0] - B Motor
##	[Y in MOTOR1] - A Motor
##	[E in MOTOR6] - Extruder
##	Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
##--------------------------------------------------------------------
##serial: /dev/serial/by-id/usb-Klipper_stm32g0b1xx_3D003F0011504B4633373520-if00
#restart_method: command
##--------------------------------------------------------------------

[printer]
kinematics: corexy
max_velocity: 300  
max_accel: 3000    			#Max 4000
max_z_velocity: 15 			#Max 15 for 12V TMC Drivers, can increase for 24V
max_z_accel: 350
square_corner_velocity: 5.0

[temperature_sensor MCU]
sensor_type: temperature_mcu

[temperature_sensor SoC]
sensor_type: temperature_host

######################### POWER CONTROLLER ###########################
[output_pin PS_ON_OFF]
pin: PC3
value: 1
#####################################################################

# ######################### BEACON SCANNER ###########################
[beacon]
 serial: /dev/serial/by-id/usb-Beacon_Beacon_RevD_45FABF8D4E4B333448202020FF0A0621-if00
x_offset: -1.719 # update with offset from nozzle on your machine
y_offset: -73.239 # update with offset from nozzle on your machine
mesh_main_direction: x
mesh_runs: 1

[bed_mesh]
speed: 420
horizontal_move_z: 12
mesh_min: 40, 30
mesh_max: 340, 280
probe_count: 6, 6
#---------------------------------------------------------------------------
[idle_timeout]
timeout: 1800

[safe_z_home]
[... 374 lines cut ...]
#*# 	  0.16611288795613285
#*# model_domain = 3.296938592977582e-07,3.3478753749769333e-07
#*# model_range = 0.200000,5.000000
#*# model_temp = 49.271195
#*# model_offset = 0.00000
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	-0.223671, -0.167909, -0.187913, -0.171787, -0.175397, -0.165027
#*# 	-0.190403, -0.144021, -0.160173, -0.156289, -0.163816, -0.147125
#*# 	-0.179681, -0.164370, -0.185062, -0.169769, -0.175827, -0.173186
#*# 	-0.212819, -0.194417, -0.220745, -0.210980, -0.215077, -0.202902
#*# 	-0.192705, -0.165309, -0.186978, -0.169542, -0.174999, -0.157046
#*# 	-0.239936, -0.194208, -0.202588, -0.177917, -0.185569, -0.167155
#*# x_count = 6
#*# y_count = 6
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*# min_x = 40.0
#*# max_x = 340.0
#*# min_y = 30.0
#*# max_y = 280.0

rafous, 2023-11-28:
I have the same problem, did you find a solution?

fish1942, 2023-11-29:
No I went back to the 2209's

BelgarionNL, 2024-05-24:
old but for anyone the config is completely wrong. for tmc5160 stepper_X etc.