bigtreetech/Manta-M8P #93: Unable to flash Manta M4P

Gabriel121217, 2023-11-07:
I am having difficulty flashing the MCU of the manta M4P. I have tried the following
- compiling klipper myself and using SD card slot (full size SD card)
- used the precompiled bin file on the github (full size SD card)
- tried using DFU mode (not sure how exactly to put into DFU mode)

I am quite stumped and not sure what else to do. It was powered by 24V bench top power supply. The CB1 i am using has been flashed and I can access the online GUI of mainsail.

fish1942, 2023-11-07:
Make sure you are using the correct SD card slot 
or try using kiah https://github.com/bigtreetech/kiauh

Gabriel121217, 2023-11-07:
i have tried flashing from both SD card slots. which one should be the correct one?

fish1942, 2023-11-09:
The one on the top of the board

alexnagelberg, 2025-01-09:
I went through two boards before I figured out what was going on. It doesn't sound like it will resolve the OPs since they used the full-size SD card, but for others, the 'hurakan' firmware that's preloaded wouldn't work with DFU flashing, and it didn't even occur to me there were two SD card slots. Loaded new klipper firmware on full size as firmware.bin with a fat32 filesystem, restarted the MCU, then was able to use DFU mode with the latest klipper on it.