bigtreetech/Manta-M8P #95: Manta M8P with CM4 not found the CSI Pi Global Shutter Camera and no working CSI Display

Magic2991, 2023-11-15:
I have problems with the CSI ports for camera and display. Neither the PI Global Shutter Camera nor the display connected to the CSI port work with the Manta M8P+CM4. With an HDMI display it works without any problems, with CSI it stays dark.

looxonline, 2023-11-22:
You need to install the overlays on the CM4 to enable the camera and the screen which has nothing to do with the manta. https://www.raspberrypi.com/documentation/computers/compute-module.html#sources

NegussugeN, 2023-11-28:
> I have problems with the CSI ports for camera and display. Neither the PI Global Shutter Camera nor the display connected to the CSI port work with the Manta M8P+CM4. With an HDMI display it works without any problems, with CSI it stays dark. I usw the pi OS Bullseye.

Did you ever make any progress on this? Having the same issue myself.

Superflyron, 2023-12-30:
I wanted to thank you very much, I tried everything possible for two days without success. Your solution worked right away, you should forward it to bigtreetech, because it is wrong in the description.

Turge08, 2023-12-31:
> I wanted to thank you very much, I tried everything possible for two days without success. Your solution worked right away, you should forward it to bigtreetech, because it is wrong in the description.

I notified them about the error. It's related to a change in the newest version of Raspbian btw (bookworm).