bigtreetech/Manta-M8P #98: Canbus marked incorrectly on the pinout diagram

Pneumanifest, 2023-12-13:
On the M8P V2 pinout diagram the canbus pins are marked as TX and RX. This is the wrong protocol and and does not translate to over to canbus High and Low. 

Recommend to update pinout and add schematic to prevent user confusion.

[image]

olivbd, 2023-12-19:
I confirm that these pins should be marked "CANH"/"CANL" (or "CANL/CANH" ?)

There is a transceiver on this board (SN65HVD1050D), you can check it on schematics.

[image]

dhazelett, 2023-12-21:
TX = CAN Low, RX = CAN High. I discovered this a while ago by flipping the board over, woulda been very helpful to have it labeled correctly in the pinout.

Pneumanifest, 2023-12-21:
Patrick Myrseth 
reacted to your message: 

 

view it on GitHub, or 
unsubscribe.

Pneumanifest, 2023-12-21:
Patrick Myrseth 
reacted to your message: