bigtreetech/Manta-M8P #103: Manta M8P V2 PWM fans not working

mrstone14, 2024-01-10:
Trying to run Noctua 12V fans off the PWM pins, but the fans never kick on.  12v jumpers are correctly installed.  I had success with the Manta M8P V1.1, but not on the V2.0.  I am getting the following error from Klipper:

"MCU 'mcu' shutdown: Not a valid PWM pin"

My fan config:

[fan_generic exhaust]
pin: PC1
max_power: 1.0
shutdown_speed: 1.0
kick_start_time: 0.5
tachometer_pin: PA2
hardware_pwm: True
cycle_time: 0.00004 # 25 kHz

If I turned hardware_pwm to "false", the klipper error goes away but the fan still doesn't run.  Per the documentation, PWM fans should be pin PC1 and PC2.

mrstone14, 2024-01-10:
Figured out the problem: 

The PWM pins and Tachometer pins on the V2 pinout jpeg are reversed.  When viewing the 3D file provided by BTT, you can see the pins labeled on the bottom of the PCB, which are swapped compared to the jpeg file.  I have reversed the pins in my Klipper config and all is working.  

This appears to be the case for Fan 4, Fan 5 and Fan 6. I have confirmed this is working for Fan 5 and Fan 6, but have not tested Fan 4.  

Posting photos to help anyone else.  Suggested Klipper config:

FAN 4 (not confirmed)
pin: PA4

FAN 5 (confirmed)
pin: PA6
tachometer_pin: PC2

FAN 6 (confirmed)
pin: PA2
tachometer_pin: PC1

[image]

[image]

closed 2024-01-10 by mrstone14

jurassicplayer, 2024-01-19:
Could probably leave this issue open, since it's still an issue with their documentation (threw me for a loop too xD).