bigtreetech/Manta-M8P #107: CAN to USB missing after flashing bootloader

garethbradley, 2024-02-06:
Board: Manata M8P v2
MCU: CB1 with eMMC

After flashing the M8P board with a new klipper config, I lost connectivity with my EBB42 v1.2. So, I reflashed the bootloader as stated here and lost the OpenMoko usb device. I'm now completely unable to boot klipper or cannect to the can0 socket

```
biqu@BTT-CB1:~$ lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
biqu@BTT-CB1:~/klipper$ python3 ~/katapult/scripts/flash_can.py -q
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/biqu/katapult/scripts/flash_can.py", line 487, in run_query
    self.cansock.bind((intf,))
OSError: [Errno 19] No such device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/biqu/katapult/scripts/flash_can.py", line 615, in main
    loop.run_until_complete(sock.run_query(intf))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/biqu/katapult/scripts/flash_can.py", line 489, in run_query
    raise FlashCanError("Unable to bind socket to can0")
FlashCanError: Unable to bind socket to can0
```

Any ideas? I've seen others saying the board is now bricked...

michmerr, 2024-04-04:
I'm having the same issue. In my case, I accidentally flashed the EBB toolhead images to the M8P, which started the problem. Flashing the M8P v2 bootloader and klipper images didn't seem to help. I get the same flash_can.py output, and the OpenMoko device is no longer in the lsusb output. I still have the can0 device config, but can0 doesn't show up in 'ip a' output, nor (predictably) can it be explicity started.

michmerr, 2024-04-04:
...and now I think I've resolved the issue.

Here's my deduced timeline.

- Flashed stock M8P with Klipper (successful)
- Flashed M8P with canboot and Klipper _intended for the EBB_ (successful flash, wrong target)
- Flashed EBB with canboot and Klipper (successful flash, validation script failed)
- Flashed M8P with V2 bootloader from this repo (successful)
- No can0 interface
- Flashed M8P with Klipper (incorrectly!)
- No can0 interface

Upon review, I _think_ that my last flash of Klipper to the M8P was misconfigured. Instead of configuring the communication interface to "USB to CAN bus bridge (USB on PA11/PA12)", I think might have set it to "CAN bus (on PD0/PD1)". Maybe. (The config was overwritten by the one for the EBB image, and I could not remember what the previous day's config had been for sure.)

I just rebuilt the Klipper image for the M8P, making sure to set the communication interface to "USB to CAN bus bridge (USB on PA11/PA12)" and the CAN bus interface (which only shows up in the config after setting the communication interface to "USB to CAN bus bridge (USB on PA11/PA12)") to "CAN bus (on PD0/PD1)".

On flash later, and can0 is visible again, and the validation scripts dump the uuids for both the M8P and EBB nodes of the CAN bus. 

Hopefully this helps anyone else that got into the same condition via the same set of mistakes.

charchess, 2024-04-05:
Hi, 
not sure about your issue but I found this about canbus : https://canbus.esoterical.online
I was struggling with can, can to bridge and so (EBB42 and M8P) and ...this is just THE guide on how to make it work, I went thoruhg this step by step and ...it just finally worked, this guide is amazing and should be put forward for anyone that want to do some CAN bus implementation, 

hope it'll help

garethbradley, 2024-04-05:
I'm glad you got it sorted @michmerr - I was just coming on here to confirm that it is fixable (at least I got it working again) but failed to keep any notes as to how!

manotroll, 2024-04-28:
I don't want to use can mode just usb standard mode

manotroll, 2024-04-28:
I found the solution to the problem in a video on YouTube, I'll put it here so you can see
There is an error in the PDF explanation in the manual on how to do it
https://www.youtube.com/watch?v=k8DmTrY2GgY

TheBinaryLoop, 2024-05-12:
None of the solutions above solved it for me. I still can't see the device as an usb device (canbus disabled). Any other helpful tips?