bigtreetech/Manta-M8P #111: Manta M8p V2 -PWM / Tach-Pins swapped-

oMTEo, 2024-03-31:
While Im on setting up PWM-Fans on my new Manta-M8P-V2 I´ve found 2 wrong pin-definitions in the manual for FAN5 and FAN6.
PWM and Tacho-pins are swapped ! (I´ve cross-checked with the schematics)

Please check and correct all files and give me some salary for hours of exploring bugs on your products :(

[image]

oMTEo, 2024-04-01:
Dear BTT, you forgot page 19...

[image]

looxonline, 2024-04-02:
Fixed

bigtreetech, 2024-04-02:
Docs updated. Thanks for pointing this out.

closed 2024-04-02 by bigtreetech