bigtreetech/Manta-M8P #112: M8p v2 + CB1 v2.2 bootloader issues; no serial connection

cribbageSTARSHIP, 2024-04-14:
Good day.

I can get the cb1 to boot but the firmware.bin is never picked up by the sdcard reader. I've tried DFU mode but it always errors out: 

`biqu@BTT-CB1:~/klipper$ sudo dfu-util -a 0 -D ~/klipper/out/klipper.bin --dfuse-address 0x08000000:force:mass-erase -d 0483:df11 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Performing mass erase, this can take a moment
Downloading to address = 0x08000000, size = 29308
Download	[=========================] 100%        29308 bytes
Download done.
File downloaded successfully
biqu@BTT-CB1:~/klipper$ make flash FLASH_DEVICE=0483:df11
  Flashing out/klipper.bin to 0483:df11
sudo dfu-util -d ,0483:df11 -R -a 0 -s 0x8020000:leave -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08020000, size = 29308
Download	[=========================] 100%        29308 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status

Failed to flash to 0483:df11: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=0483:df11

make: *** [src/stm32/Makefile:111: flash] Error 255
`

manotroll, 2024-04-28:
It even records but even restarting it doesn't exit UFD mode

manotroll, 2024-04-28:
I found the solution to the problem in a video on YouTube, I'll put it here so you can see
There is an error in the PDF explanation in the manual on how to do it
https://www.youtube.com/watch?v=k8DmTrY2GgY

Ivzo, 2024-05-16:
same, no serial after flashing bootloader.

MBOKjr, 2024-06-26:
Same here, I even tried what the video showed to do that @manotroll posted.

I'm using CANBus so the UUID took it's place, but the fact that we still can't get it persists.

royalfifty8, 2024-06-27:
Also having issue. Video posted by @manotroll helps, kinda. But does not solve the issue. Tried reaching out to BIGTREETECH but never got a reply.

lavidaesrecorta, 2024-12-17:
I know I'm kinda late but I had the same problem. It was the CB1 not well seated.

I realized this when I was not able to see any USB serial, so I gave a little tap to the CB1 on the side next to the mini-HDMI ports and dmesg showed that I had connected (and immediatly disconnected) varoius USB devices. 

Use `dmesg -HW` and connect a USB device to the M8P ports, if nothing shows you might have to re-seat the board. 

After re-seated, I could still not see the `/dev/serial` directory, but I could see `Terminus Technology Inc. Hub` using `lsusb` (It was not available before re-seating). If you see that, you can follow the method shared by @manotroll.

Because the video is in german and I don't want people to fry their boards, here is what you have to do (Or you can watch the [video](https://youtu.be/k8DmTrY2GgY?si=LaNvqh3wWt9HQ9PF&t=463) and follow like me, he does this at 7:45):

1.- Put the MCU in DFU mode (Hold BOOT0 and tap the RESET). You may want to use `dmesg -HW` in the CB1 via SSH before doing so, that way you can see the MCU connecting in DFU mode to the CB1

2.- Get the Internal Flash Address using `dfu-util -l`. This was my output:
`
Found DFU: [0483:df11] ver=0200, devnum=3, cfg=1, intf=0, path="2-1.4", alt=1, name="@Option Bytes   /0x5200201C/01*88 e", serial="364A37693331"
Found DFU: [0483:df11] ver=0200, devnum=3, cfg=1, intf=0, path="2-1.4", alt=0, name="@Internal Flash   /0x08000000/8*128Kg", serial="364A37693331"
`
As you can see, the Internal Flash Address is 0x08000000.

3.- Flash the board using the internal address that you got in the previous step:
`dfu-util -R -a 0 -s 0x08000000:mass-erase:force -D ~/klipper/out/klipper.bin`

4.- Power off and on again, after the reset you should be able to see the /dev/serial directory.

NOTE: After flashing and resetting, I get this new device in `lsusb`:
`Bus 002 Device 003: ID 1d50:614e OpenMoko, Inc. stm32h723xx`

If you see this device and still not seeing the /dev/serial directory, it may be an OS problem. It was not my case.

Good luck to everybody!

hiddengerbil, 2025-03-08:
> I found the solution to the problem in a video on YouTube, I'll put it here so you can see There is an error in the PDF explanation in the manual on how to do it https://www.youtube.com/watch?v=k8DmTrY2GgY

I don't speak any German, so this can't help me. Any English landguage solutions?

DaveuRrr, 2025-03-14:
> 1.- Put the MCU in DFU mode (Hold BOOT0 and tap the RESET). You may want to use `dmesg -HW` in the CB1 via SSH before doing so, that way you can see the MCU connecting in DFU mode to the CB1
>
> 2.- [...]

I found myself in the same boat and followed the same steps as mentioned above but for me it did not work. The video was helpful since it mentions `dmesg -HW` and you can see your Manta board go into DFU mode.

My Manta was previous working but after doing maintenance to it (changing out the nozzle) it all of a sudden can't see the MCU. Is there a way to test to see if whatever communication method the Manta is using to communicate works? Flashing to DFU Mode seems like it's working but I still can't see it via `ls /dev/serial/by-id/*`

Not sure what else to do here, I don't want to buy a new board and end up with the same issue. :(

```linux
daveurrr@mainsail:~ $ sudo dfu-util -a 0 -D ./M8P_bootloader.bin --dfuse-address 0x08000000:force:mass-erase -d 0483:df11
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Performing mass erase, this can take a moment
Downloading to address = 0x08000000, size = 8064
Download        [=========================] 100%         8064 bytes
Download done.
File downloaded successfully
```

DaveuRrr, 2025-03-30:
I was able to get my M8P to work after following [Manta M8P & BTT EBB SB22XX](https://forum.vorondesign.com/threads/manta-m8p-btt-ebb-sb22xx-setup.1697/)

Since I am using CAN BUS, I need to flash my EBB36 and my Manta M8P but I am also using a U2C. When looking at the instructions and referencing back from what I remember, I only flashed the Manta M8P once when I set up CAN BUS for the first time with canboot. I flashed not really knowing what I was doing but following the thread mentioned above I was able to get my Manta working again. We are flashing in 2 spots. `0x08000000` which is for katapult / canboot and `0x08002000` is for klipper on the Manta. Once I did that it all started working again. It would be great if tutorials would explain what the commands are doing. Most of us are just copy & pasta'ing code / commands and expecting things to just work without understanding the context of the command. Oh well.

I hope someone finds the link useful as it worked for me.