bigtreetech/Manta-M8P #113: m8p / cb1   extruder not working

lkelley52, 2024-04-23:
can not get the extruder to work. if i plug the extruder motor in to one of the axis’s motor it moves . so motor is good. i have the extruder plugged into motor 5 on the board. i have checked wiring. no breaks and wired correctly, board pins good. 

Motor5
[extruder]
step_pin: PG13
dir_pin: PG12
enable_pin: !PG15
microsteps: 16
rotation_distance: 33.500
full_steps_per_rotation:200
nozzle_diameter: 0.4
filament_diameter: 1.75
heater_pin: PA0 # HE0
sensor_pin: PB0 # T0
sensor_type: Generic 3950
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

[klippy (3).log](https://github.com/bigtreetech/Manta-M8P/files/15082675/klippy.3.log)

Seriy2103, 2024-11-02:
I have same issue. Stepper motor and wiring are good but on M5-M8 extruder not working. Any one have some solutions ?

royalfifty8, 2024-11-02:
Have you done your configuration checks? Looking at klippy log, it looks like your extruder is not hot enough. As a safety feature Klipper wont allow extruder to work below a certain temp. You can check to see that the motor is working by verifying your tepper motors. Use STEPPER_BUZZ command. For extruder it will be (STEPPER=extruder). Make sure there is no filament loaded before doing so. Otherwise ring up your heater to extruder temps and then try extruding some material.b

royalfifty8, 2024-11-02:
@Seriy2103  When setting up Klipper. This step is important, as it allows you to safely verify everything is working without damaging your printer and/or its components. 
https://www.klipper3d.org/Config_checks.html

Seriy2103, 2024-11-04:
I did everything as described in the guide, nothing helped. Most likely, something is wrong with the pins step_pin, dir_pin and enable_pin. Everything else works correctly.

rhiz0matic, 2025-05-26:
Same issue but with cb2 has anyone figured it out?

rhiz0matic, 2025-05-26:
I figured it out

There are 8 stepper motor drivers 
BUT there are 9 stepper motor ports/plugs for a stepper z/1 on driver 3

So to correct it, corresponding stepper motor plugs go
- Plug 1 Driver 1 Motor 1 Stepper x
- Plug 2 Driver 2 Motor 2 Stepper y
- Plug 3 Driver 3 Motor 3a Stepper z
- Plug 4 Driver 3 Motor 3b Stepper z1
- Plug 5 (NA)     Motor 4 (Stepper NA)
- Plug 6 Driver 5 Motor 5 Stepper Extruder

So I plugged my Motor 5 into Plug 5 but it should go into Plug 6