bigtreetech/Manta-M8P #124: New Manta M8p v2.0 CAN not working

tekknix, 2024-08-08:
Hi,

i've exchanged my LDO Leviathan v1.2 Board against the Manta M8p V2.0

Before the Leviathan i've had running a Manta M8p v1.0 and U2c V2.1 and an EBB36 v1.2 via CAN.

Flashing the Manta with Katapult and Klipper was no problem. The Can-Bridge is working and i can see the UUID of the Manta.

I've tried to connect the EBB36 flashed with Katapult, but the Manta will not detect the Toolhead-Board. 

I've tried to flash the EBB36 PreCompiled Firmware but this leads to the Problem, that after Connecting the CAN-Bus UUID of the Manta will also not shown up.

Output of LSUSB:

```
Bus 001 Device 004: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 003: ID 04d8:e72b Microchip Technology, Inc. Beacon RevH
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Out of canbus-Query:

```
tekknix@Reborn2Pro:~ $ ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
Total 0 uuids found
```

Directly after PowerOn i can see the UUID at least from Manta but after a few seconds it's gone:

```
Last login: Thu Aug  8 14:03:26 2024 from 192.168.1.12
tekknix@Reborn2Pro:~ $ python3 ~/klipper/lib/canboot/flash_can.py -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 212fadcb00b4, Application: Klipper
Query Complete
tekknix@Reborn2Pro:~ $ python3 ~/klipper/lib/canboot/flash_can.py -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Detected UUID: 212fadcb00b4, Application: Klipper
Query Complete
tekknix@Reborn2Pro:~ $ python3 ~/klipper/lib/canboot/flash_can.py -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Query Complete
tekknix@Reborn2Pro:~ $ python3 ~/klipper/lib/canboot/flash_can.py -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Query Complete
```

Any help i appreciate.

Regards
MoS-tekknix

bigtreetech, 2024-08-08:
UUIDs will disappear as soon as the klipper service uses them. This is documented in the klipper documentation.

If you have these UUIDs listed in the cfg file then they will disappear. They will still be accessible by the katapult flashtool though.

tekknix, 2024-08-08:
Please don't close Issues which are not solved. That's not the Idea.

Anyhow, i know how to switch off klipper before trying to list my CAN-Bus Devices (sudo systemctl stop klipper.service)

And additional without getting an UUID of the Board i'm not able to flash it via katapult, cause it can't reset to bootloader.

```
tekknix@Reborn2Pro:~/klipper $ python3 ~/klipper/lib/canboot/flash_can.py -q
Resetting all bootloader node IDs...
Checking for canboot nodes...
Query Complete
```

`~/klippy-env/bin/python ~/katapult/scripts/flashtool.py -i can0 -u 212fadcb00b4 -r
`

Where 212fadcb00b4 is the uuid the board offers directly after restart before it's gone away.

Therefore either the CAN-Mapping of PD0/PD1 as CAN - PINs is wrong for my Manta M8p V2.0 or the CAN isn't working at all.

Regards
MoS-tekknix

bigtreetech, 2024-08-08:
> Please don't close Issues which are not solved. That's not the Idea.
>
> Anyhow, i know how to switch off klipper before trying to list my CAN-Bus Devices (sudo systemctl stop klipper.service)
>
> And [...]

CAN is working by virtue of the fact that it appears after boot. Even stopping the klipper service will not cause it to reappear. Once it has been consumed, it is consumed. You need to remove it from the klipper config and reboot.

tekknix, 2024-08-08:
Thanks for Reopening this Issue. 

I've removed the MCU-Entries completely from my printer.cfg but the issue will be the same.

Maybe you've an additional idea? 

I've meassure the correct wiring from my Can-Cable, everything is right. As soon as i disconnect the EBB36, the UUID of the Manta will not disappear.

And i've 2 working EBB36 to test with. I've tried both of them, Flashed the orignal Firmware for CAN-Usage (Firmware_can.bin) via STM32.

bigtreetech, 2024-08-08:
Ensure that you have your termination resistors correctly installed.

tekknix, 2024-08-08:
I've set the Jumper on EBB36 and also the Manta

bigtreetech, 2024-08-08:
Running through a CAN bridge can be tricky. You can't program your motherboard directly using Katapult because it will register as a USB device as soon as it enters the bootloader. I have also noticed that even using the -r command parameter sometimes results in the board jumping straight back into the application as soon as it enters katapult and the only way I can get new firmware onto it is by using DFU mode.

As for the UUIDs disappearing when you plug in the EBB, that does sound as if they are being found and consumed. Try removing klipper completely using KIAUH (saving your configs), rebooting and then seeing if the UUIDs can be seen.

Also ensure that you used the correct communication settings for both katapult and the main firmware on the EBB and the mainboard (Mainboard should be USB in katapult and CAN in the main klipper fw. EBB should be CAN in both). Ensure that the buffer is 1024 in length within the linux config file and that the baud in the config file matches the baud in all firmware config.

Aside from those points, there could be a number of unique implementation issues. Github is not an official support channel for our products. It is a place where we monitor for firmware bugs, feature requests and confirmed hardware issues. For support please contact support003@biqu3d.com and quote this link.

closed 2024-08-08 by bigtreetech

tekknix, 2024-08-08:
Hi, unfortunately i'm getting back an Error-Message that this Email-Adresse either does not exists or that i'm not allowed to send something to you:

host mxbiz1.qq.com [103.86.78.3:25] said: Mailbox unavailable or access 
              denied 
              [MM26xUelqNM/GlHlV5vGWiq4eeKDisoOC/0PK741IdkZHDYFes8lK5M= 
               IP: 81.169.146.217]. 
              https://open.work.weixin.qq.com/help2/pc/20057.

bigtreetech, 2024-08-08:
> Hi, unfortunately i'm getting back an Error-Message that this Email-Adresse either does not exists or that i'm not allowed to send something to you:
>
> host mxbiz1.qq.com [103.86.78.3:25] said: [...]

Pop a ticket off. It's easier than email: https://biqu3d.com/pages/submit-a-ticket

tekknix, 2024-08-08:
I like to try, but

[image]

bigtreetech, 2024-08-08:
Doh! I'll get the ticketing system fixed as soon as people are in at the office tomorrow morning.

bsivley, 2025-01-13:
was there ever a resolution to this IM having a very similar issue

looxonline, 2025-01-13:
> was there ever a resolution to this IM having a very similar issue

If you follow the thread you'll notice that the UUIDs disappearing was not a flaw but rather how the functionality is supposed to work. The Klipper and Katapult are great resources for this kind of documentation.